micro-ROS/micro_ros_arduino #1491: undefined reference to `__locale_ctype_ptr' issue

ChoiYouJung, 2023-08-03:
## Issue template

- Hardware description:  i.MX-RT1060 MCU
- RTOS:  No
- Installation type:  MCUXpresso IDE
- Version or commit hash:  humble

#### Steps to reproduce the issue

(1) Work in progress(purpose)
I want to operate the micro-ros-arduino library on a non-official board in MCUXpresso IDE.

(2) Code Behavior Description
* Creating 1 publishers and 1 timers on 1 node to publish ros message
* The code is written based on the examples in micro_ros_arduino/examples/micro-ros_publisher.
* Uses precompiled libraries from Teensy 4.1, a supported board.

#### Expected behavior
* Build works fine.

#### Actual behavior
* The build fails with the following error
* error content
```
Invoking: MCU C++ Linker
arm-none-eabi-c++ -nostdlib ........

c:/nxp/mcuxpressoide_11.4.0_6237/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.4.0.202103011116/tools/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe:
D:\mros_ws\project\lib\libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'

c:/nxp/mcuxpressoide_11.4.0_6237/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.4.0.202103011116/tools/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe:
D:\mros_ws\project\lib\libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xbe): undefined reference to `__locale_ctype_ptr'

c:/nxp/mcuxpressoide_11.4.0_6237/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.4.0.202103011116/tools/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: 
D:\mros_ws\project\lib\libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x46): undefined reference to `__locale_ctype_ptr'

c:/nxp/mcuxpressoide_11.4.0_6237/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.4.0.202103011116/tools/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: validate_topic_name.c:(.text.rcl_validate_topic_name+0xb6): undefined reference to `__locale_ctype_ptr'
c:/nxp/mcuxpressoide_11.4.0_6237/ide/plugins/com.nxp.mcuxpresso.tools.win32_11.4.0.202103011116/tools/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld.exe: validate_topic_name.c:(.text.rcl_validate_topic_name+0x164): undefined reference to `__locale_ctype_ptr'
```

#### Additional information
* The operating system of the MCUXpresso IDE is Windows 10.

* What else have you tried:
Build by adding the files with errors (validate_topic_name.c, validate_full_topic_name.c, validate_node_name.c) to the folder

* On further attempts, the build succeeded.

* Question: What should I do to fix that error? Or what could be causing that error?

pablogs9, 2023-08-03:
Ensure that you build your code using a compiler that is compatible with the one used for building your micro-ROS static library.

ChoiYouJung, 2023-08-08:
Thanks for the answer.

**How to check the compiler version of micro-ros-arduino static library?**

For reference, the version of ros2 I'm using is Humble.

pablogs9, 2023-08-08:
Here you can see the docker that we use for building this static libraries, and also you can found the teensy toolchain used: https://github.com/micro-ROS/docker/blob/b6c0c8fc7e6cfabb64746d57f5d25da43fb8880e/micro-ROS-static-library-builder/Dockerfile#L12

Please take a look and let me know if you solve the issue.

ChoiYouJung, 2023-08-09:
I looked at the micro_ros_arduino/extras/library_generation/library_generation.sh you pointed out. So I have an additional question.

**In the micro_ros_arduino/extras/library_generation/library_generation.sh file, line 167 has no information about the compiler version.** 

**For reference, I tried moving to the uros_ws/teensy-compile/tools/arm/bin/arm-none-eabi- folder path, but I couldn't get the version information of the compiler.**

https://github.com/micro-ROS/micro_ros_arduino/blob/ea286412a524cb7c0eec5a2f6f92ae283b9bd0df/extras/library_generation/library_generation.sh#L167

**If so, is the version like gcc-arm-none-eabi-5_4-2016q3 shown on line 167 in the address below correct?**

https://github.com/micro-ROS/micro_ros_arduino/blob/foxy/extras/library_generation/library_generation.sh

https://github.com/micro-ROS/micro_ros_arduino/blob/3a187e551e0e94d469d6016937f482057fa56631/extras/library_generation/library_generation.sh#L167

**And for reference, I found out the compiler version I'm using in the MCUXpresso IDE, which is the environment I'm developing on.**

**arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.2.1 20201103 (release)**

ChoiYouJung, 2023-09-07:
> Here you can see the docker that we use for building this static libraries: https://github.com/micro-ROS/docker/blob/b6c0c8fc7e6cfabb64746d57f5d25da43fb8880e/micro-ROS-static-library-builder/Dockerfile#L12
> 
> And here are the toolchains used:
> 
> https://github.com/micro-ROS/micro_ros_arduino/blob/ea286412a524cb7c0eec5a2f6f92ae283b9bd0df/extras/library_generation/library_generation.sh#L65
> 
> Please take a look and let me know if you solve the issue.

Hello.

As you advised, I created a new library by adding the following to the Dockerfile.

https://github.com/micro-ROS/docker/blob/humble/micro-ROS-static-library-builder/Dockerfile
micro-ROS-static-library-builder/Dockerfile

```
    && wget --no-check-certificate https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2 \
    && tar -xvf gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2 \
    && rm -rf gcc-arm-none-eabi-10.3-2021.10-x86_64-linux.tar.bz2 gcc-arm-none-eabi-10.3-2021.10/share/doc
```
The library was created successfully.
When I built the created library in MCUXpresso IDE, the error mentioned at the beginning disappeared and the build was successful.

thank you.

When I run the result of a successful build on my development board, there is a problem with the MCU being reset in the following code.

```
rclc_support_init(&support, 0, NULL, &allocator);
```

The screen capture of the agent is as follows.
[image]

**As a result, I have a question. Does anyone know the cause or solution to the issue with the code above?**

ChoiYouJung, 2023-10-10:
Hello.
The additional issues I asked about above were resolved well.
The above issue was an MCU reset due to watchdog timeout.
I increased the watchdog's timeout and it worked well.

It works well on the i.MX-RT1060 MCU development board.

thank you.

closed 2023-10-10 by ChoiYouJung