micro-ROS/micro_ros_arduino #1494: Arduino Zero: rclc_support_init hangs

MarkFloatingInSpace, 2023-08-05:
- Hardware description: Arduino Zero
- Installation type: Native Ubuntu 22.04 with ROS2 Humble
- Version or commit hash: Humble

The micro-ros_publisher example hangs at the rclc_support_init function.

Output:
```
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6
[1691262472.164089] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1691262472.164303] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6
[1691262474.078448] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1691262474.088567] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1691262475.097059] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1691262475.210031] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
```

pablogs9, 2023-08-07:
Hello @MarkFloatingInSpace, micro-ROS Humble is deprecated, could you give Iron a try?

MarkFloatingInSpace, 2023-08-07:
Hello Pablo,
 
thank you for your response! I appreciate it!

Unfortunately I observe the same problem when using Iron.

I deleted the Humble Arduino lib and installed the Iron lib. I tested the Iron agent in a docker container. Passing the serial port both by name and by device id:

sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:iron serial --dev /dev/serial/by-id/usb-Arduino_LLC_Arduino_Zero_5B6C7B8050583153382E3120FF010720-if00 -v6

and

sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:iron serial --dev /dev/ttyACM0 -v6
 
The output was the same as in my original post. I altered the example publisher code by creating a second error_loop() function with a different LED blinking timing. Therefore I know that the rclc_support_init function fails.
 
Do you have another suggenstion?
 
Thanks again for your support!

Yours,
Markus

pablogs9, 2023-08-08:
Hi Markus, I've no Zero available for tests bc it is a community-contributed board, so I guess that I cannot help you too much...

It seems in your log that the board disconnects, and it is kind of weird because AFAIK Arduino Zero has a dedicated USB-Serial chip so it should not disconnect automatically. Have you tried with another USB cable?

Also, this MCU has a very small RAM memory, and recently micro-ROS have increased slightly its memory usage, just for testing, does it behaves equally if you used the [deprecated foxy version](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.5-foxy)?

MarkFloatingInSpace, 2023-08-09:
Hi Pablo,
 
I understand of course that a community supported board comes with no guarantees! Thank you for your help nonetheless!

I would like to note that the board does NOT disconnect! I just reset the board to make sure that the agent is running when the board initializes. That is what causes the "Serial port not found" during the restart of the board. I am sorry for having caused this confusion! Does this change your diagnosis maybe?

I have tried both a different cable and the deprecated foxy version, both with no luck.
 
Yours,
Markus

pablogs9, 2023-08-10:
Not really, it would be interesting if you can debug your code to see where exactly is the board hanging. If you find a slot for doing so, please let us know, that would provide details about the problem here.

MarkFloatingInSpace, 2023-08-10:
I found it! I was using the native USB port instead of the programming/debugging USB port. Sorry for this stupid mistake on my part! Everything works now.
 
Thank you for your support and for maintaining such an awesome library!

closed 2023-08-10 by pablogs9