micro-ROS/micro_ros_arduino #1498: Q: NTP rmw_uros_sync_session usage

dhood, 2023-08-15:
Hi maintainers; high-level question for you:

In general, with what frequency would you expect `rmw_uros_sync_session` to be called, relative to e.g. `rmw_uros_epoch_millis`? I understand from the uros and [xrce docs](https://micro-xrce-dds.docs.eprosima.com/en/stable/client_api.html) that the synchronisation needs to be done before the timestamp access, but from other NTP usage I would expect that synchronisation needs to be done at a much lower frequency than the timestamps are being accessed.

Any general guidance/rules of thumb?

Thanks in advance

pablogs9, 2023-08-16:
Hello @dhood, we do not have any recommended value or rule of thumb. But as you mention the sync can be slower than the access. The synchronization needs to be done before (at least once) timestamp access.

AFAIK the frequencies shall be related to the drift you expect from your clocks. In an ideal scenario, with no clock drift, a single synchronization is enough. The more drift, the more you need to resync.

dhood, 2023-10-13:
Thanks for the response!

closed 2023-10-13 by dhood