micro-ROS/micro_ros_arduino #1503: Publisher basic example does not work in Teensy 4.1 after receiving a new docker image micro_ros_static_library_builder Pull

ChoiYouJung, 2023-08-24:
## Issue template

- Hardware description:  Teensy4.1
- RTOS:  No
- Installation type:  Arduino IDE (Version : 2.1.0)
- Version or commit hash:  humble

#### Steps to reproduce the issue

(1) code behavior
* Create 1 Publisher and 1 Timer on 1 node to publish ros message
* Using examples from micro_ros_arduino/examples/micro-ros_publisher

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 0;
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_so
```

#### Expected behavior
* 1 timer publishes a ROS message every second.

#### Actual behavior
* Confirm that 1 publisher is created.
* The message is not published.

#### Additional information
* The operating system of the Arduino IDE is Windows 10.

* In WSL, the precompiled library was built by referring to the following address.
**- Reference address 1 : https://github.com/micro-ROS/micro_ros_arduino**
**- Reference address 2 : https://hub.docker.com/r/microros/micro_ros_static_library_builder**

* After updating the microros/micro_ros_static_library_builder Docker image using the Docker Pull command around 23.08.17, the code does not work.

* The existing image was used around 23.06.29 using the Docker Pull command.

* Question: Are there any changes to microros/micro_ros_static_library_builder Docker between 23.06.29 and 23.08.17? Also, is there any cause of the problem that you know about this issue?

pablogs9, 2023-09-04:
Teensy support was updated to match the latest Arduino Teensy Support: 

- https://github.com/micro-ROS/docker/pull/97
- https://github.com/micro-ROS/micro_ros_arduino/pull/1482

Are you using the latest Arduino Teensy support?

ChoiYouJung, 2023-09-08:
> Teensy support was updated to match the latest Arduino Teensy Support:
> 
> * [Update teensy compiler docker#97](https://github.com/micro-ROS/docker/pull/97)
> * [Fix teensy patch and compiler #1482](https://github.com/micro-ROS/micro_ros_arduino/pull/1482)
> 
> Are you using the latest Arduino Teensy support?

Thanks for your help.
When I tested the micro-ros_publisher and micro-ros_reconnection_example examples on the Teensy 4.1 development board as you advised, they worked successfully.

But I have one additional question.
There is a code written in the Arduino IDE that creates and operates 16 Publishers and 8 Subscriptions on one node.
It worked normally in teensy 1.57.2, but when it ran in teensy 1.58.x, a USB error occurred and publisher, subscription, and node were not created during agent reconnection. Have you ever experienced this problem?

pablogs9, 2023-09-08:
Are you rebuilding the library for allowing 16 publishers and 8 subs in the colcon meta?

ChoiYouJung, 2023-09-11:
> Are you rebuilding the library for allowing 16 publishers and 8 subs in the colcon meta?

that's right.
I rebuilt the library by creating a colocon.meta file as follows.

```
{
    "names": {
        "tracetools": {
            "cmake-args": [
                "-DTRACETOOLS_DISABLED=ON",
                "-DTRACETOOLS_STATUS_CHECKING_TOOL=OFF"
            ]
        },
        "rosidl_typesupport": {
            "cmake-args": [
                "-DROSIDL_TYPESUPPORT_SINGLE_TYPESUPPORT=ON"
            ]
        },
        "rcl": {
            "cmake-args": [
                "-DBUILD_TESTING=OFF",
                "-DRCL_COMMAND_LINE_ENABLED=OFF",
                "-DRCL_LOGGING_ENABLED=OFF"
            ]
        }, 
        "rcutils": {
            "cmake-args": [
                "-DENABLE_TESTING=OFF",
                "-DRCUTILS_NO_FILESYSTEM=ON",
                "-DRCUTILS_NO_THREAD_SUPPORT=ON",
                "-DRCUTILS_NO_64_ATOMIC=ON",
                "-DRCUTILS_AVOID_DYNAMIC_ALLOCATION=ON"
            ]
        },
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=16",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=8",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=8",
                "-DRMW_UXRCE_TRANSPORT=custom",
                "-DRMW_UXRCE_MAX_GUARD_CONDITION=10"
            ]
        }
    }
}

```

pablogs9, 2023-09-11:
So, we need some more details about your code and approach to find out why it is not working.