micro-ROS/micro_ros_arduino #1504: New install won't upload program to Teensy

Russ76, 2023-08-29:
## Issue template

- Hardware description: 
- Celeron mini desktop 4 core unit, new Jammy install, new Ros2 Humble install
- RTOS: 
- Installed Arduino IDE 2.1.1 and added Teensyduino.
- Installation type: 
- MicroROS runs OK by itself, shows trying to connect to microcontroller
- Added newest Iron library and added library to sketch, it installed OK.
- Version or commit hash: 
- Replaced the Platform.txt file as explained in MicroROS documentation.

#### Steps to reproduce the issue

#### Expected behavior
Expected it would compile the example publisher program.

#### Actual behavior
Couldn't process the publisher program, came up with errors, included below.

#### Additional information
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/jeff/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
/home/jeff/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/jeff/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xbe): undefined reference to `__locale_ctype_ptr'
/home/jeff/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/jeff/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/jeff/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/jeff/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x46): undefined reference to `__locale_ctype_ptr'
/home/jeff/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xb6): undefined reference to `__locale_ctype_ptr'
/home/jeff/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x164): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Russ76, 2023-09-04:
We finally got this setup to work! The updates have helped. Now the script uploads to the Teensy fine. However, you threw us a curve when you called the topic "micro_ros_arduino_twist_subscriber" instead of "cmd_vel" like it always was before! Ros2 topics said it was connected but Ros2 Doctor said it wasn't connected!

pablogs9, 2023-09-04:
Closing as solved

closed 2023-09-04 by pablogs9