micro-ROS/micro_ros_arduino #1505: Inconsistent communication between micro-ROS and ROS2

nickbobescobar, 2023-08-29:
## Issue template

- Computer: Ubuntu 22.04 / ROS2 Humble
- Board: Teensy 4.1 / Arduino IDE 2.1.1 / Teensyduino 1.57.2
- Installation type: micro_ros_setup
- Version or commit hash: micro-ROS Humble

#### Steps to reproduce the issue
I have followed [this tutorial](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/) without using Docker, and using the software versions listed above.

#### Expected behavior
After uploading the micro-ros-publisher and starting micro-ros-agent from terminal, I expect the following output (copied from tutorial):
[image]

Additionally, `ros2 node list` and `ros2 topic list` should show the /micro_ros_arduino_node and /micro_ros_arduino_node_publisher, respectively.

#### Actual behavior
It seems random how far the output gets:
```
robot@robot:~/auto-table_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1693259368.446882] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693259368.447123] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1693259407.550516] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1693259407.560693] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259408.566835] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259409.569079] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259410.572228] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259411.277705] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693259416.075118] info     | Root.cpp           | create_client            | create                 | client_key: 0x49E910CB, session_id: 0x81
[1693259416.075206] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x49E910CB, address: 0
[1693259416.296727] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49E910CB, participant_id: 0x000(1)
close
^C[ros2run]: Interrupt
robot@robot:~/auto-table_ws$ ^C
robot@robot:~/auto-table_ws$ ^C
robot@robot:~/auto-table_ws$ ^C
robot@robot:~/auto-table_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1693259809.813369] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693259809.813598] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
r^C[ros2run]: Interrupt
robot@robot:~/auto-table_ws$ ^C
robot@robot:~/auto-table_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1693259838.351715] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693259838.351953] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1693259855.015757] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1693259855.025906] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259856.032274] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259857.034031] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259858.043362] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259859.046385] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259860.049477] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693259860.152613] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693259866.947794] info     | Root.cpp           | create_client            | create                 | client_key: 0x49E910CB, session_id: 0x81
[1693259866.947865] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x49E910CB, address: 0
[1693259866.962000] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49E910CB, participant_id: 0x000(1)
[1693340307.130292] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1693340307.130608] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693340307.293849] info     | SessionManager.hpp | establish_session        | session re-established | client_key: 0x49E910CB, address: 0
[1693340307.295148] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x49E910CB, topic_id: 0x000(2), participant_id: 0x000(1)
[1693340307.295463] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x49E910CB, publisher_id: 0x000(3), participant_id: 0x000(1)
[1693340307.296412] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x49E910CB, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1693340330.631863] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1693340330.645169] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693340331.133306] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
^C[ros2run]: Interrupt
robot@robot:~/auto-table_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1693340517.643620] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693340517.643942] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1693340554.516390] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1693340554.526556] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693340555.014358] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
^C[ros2run]: Interrupt
robot@robot:~/auto-table_ws$ ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1693342278.955998] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1693342278.956246] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1693342345.769066] info     | TermiosAgentLinux.cpp | fini                     | server stopped         | fd: 3
[1693342345.779225] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693342346.780496] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693342347.782638] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693342348.784084] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693342349.788619] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[1693342350.792543] info     | TermiosAgentLinux.cpp | init                     | Serial port not found. | device: /dev/ttyACM0, error 2, waiting for connection...
[... cut ...]
[1693342357.084950] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x49E910CB, participant_id: 0x000(1)
^C[ros2run]: Interrupt
```
When the output says `participant created`, then I can see the node. When it gets to `topic created`, I can see the topic. However, there doesn't seem to be any pattern indicating how far it will go.

#### Additional information
I have attempted to use Teensyduino 1.58.1 but got [this issue](https://forum.pjrc.com/threads/73230-Issues-using-Micro-ROS-s-precompiled-libraries-with-Arduino-and-Teensy) and #1378.

pablogs9, 2023-09-04:
Could you detail your build procedure, the build log and the micro-ROS Agent log using the flag `-v6`?

stevejohn16, 2023-11-29:
I'm stuck with the same issue! 
I upload my code onto ESP32 and then try to run mircoros but it doesn't detect ESP32. 
It stops at logger setup
[image]
I'm using a docker!