micro-ROS/micro_ros_arduino #1508: Support class member function for callbak of subscriber

asukiaaa, 2023-09-02:
Is there any way to use class member function for callback of subscriber?
If there is not now, do you plan to support it?

I'm going to replace ros1 rosserial to ros2 microros.
The ros1 rosserial supports registering class member function as callback of subscriber so I use classes to manage rosserial.
https://answers.ros.org/question/258249/publisher-or-subscriber-inside-a-class-rosserial/
https://github.com/ros-drivers/rosserial/blob/c169ae2173dcfda7cee567d64beae45198459400/rosserial_client/src/ros_lib/ros/subscriber.h#L66
If microros does not support it, complicated change is needed for my project.

Could you give me some advice or future plan?

pablogs9, 2023-09-04:
@JanStaschulat do you know any way of adding an object method as callback for RCLC sub?

asukiaaa, 2023-11-06:
Is there any news or plan?

Zherokt, 2024-04-22:
You just need to make the functions public members of your class and declare them static.

Then you can pass them to the add subscription function as &MyClass::callback.