micro-ROS/micro_ros_arduino #1523: Adding custom message to Teensy micro-ROS using the docker image fails

mgeppert, 2023-09-09:
## Issue template

- Hardware description: Teensy 4.1 
- RTOS: N/A
- Installation type:  VM running ubuntu 22.04.  ROS2 iron, micro ROS 2.0.7, Arduino IDE 2.2.1, Teensy Board version 1.57.2 (the latest did not work for me).  running zsh.
- Version or commit hash: iron

#### Steps to reproduce the issue
Following the procedures: `https://github.com/micro-ROS/micro_ros_arduino/tree/iron#how-to-build-the-precompiled-library` to add a custom message to the system and recompile the library.
I get this step:
`docker run -it --rm -v /home/mgeppert/Arduino/libraries/micro_ros_arduino-2.0.7-iron:/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:iron -p teensy4`

#### Expected behavior
The build to be clean and useable in the Arduino IDE

#### Actual behavior
The image runs to the following point:

`=== ./control_msgs (git) ===
Cloning into '.'...
find: cannot delete ‘/project/src/’: Directory not empty
Crosscompiled environment: cleaning path
Building firmware for generate_lib platform generic
Using provided meta: /project/extras/library_generation/colcon.meta
Starting >>> rcutils 
Starting >>> rosidl_cli
Starting >>> rosidl_typesupport_interface
Starting >>> microcdr
Starting >>> rmw_implementation_cmake
Starting >>> tracetools
Starting >>> tracetools_trace
Starting >>> test_interface_files
--- stderr: microcdr                                                                                                                                 
CMake Error at cmake/SuperBuild.cmake:19 (enable_language):
  The CMAKE_C_COMPILER:
    /uros_ws/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc
  is not a full path to an existing compiler tool.
  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:54 (include)
---
Failed   <<< microcdr [0.51s, exited with code 1]
Aborted  <<< rosidl_cli [0.53s]
Aborted  <<< tracetools_trace [0.49s]
Aborted  <<< rcutils [0.55s]
Aborted  <<< tracetools [0.49s]
Aborted  <<< rmw_implementation_cmake [0.65s]                                                                                                                        
Aborted  <<< rosidl_typesupport_interface [0.68s]
Aborted  <<< test_interface_files [0.80s]                            

Summary: 0 packages finished [1.28s]
  1 package failed: microcdr
  7 packages aborted: rcutils rmw_implementation_cmake rosidl_cli rosidl_typesupport_interface test_interface_files tracetools tracetools_trace
  3 packages had stderr output: microcdr rcutils tracetools
  70 packages not processed
[ros2run]: Process exited with failure 1
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
0 upgraded, 0 newly installed, 0 to remove and 191 not upgraded.
rsync: [sender] change_dir "/project/src/action_msgs/action_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
rsync: [sender] change_dir "/project/src/actionlib_msgs/actionlib_msgs" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1338) [sender=3.2.7]
...`

#### Additional information
This is my first experience with Docker and don't know how to find out what it is trying to do.
Prior to creating the custom messages things where working well.  So the static libraries seem to work well for what I was testing.
The Host ROS Agent sees the new message correctly and compiled just fine.

I have tried to run the docker command both as non sudo and sudo.  I even sudo su and tried to run it.  Same errors.

pablogs9, 2023-09-11:
Found no problem, cannot replicate: https://asciinema.org/a/AurgtciFlcFTkzAaDn0oomqx5

TheHassanShahzad, 2024-08-07:
ROS2 humble
Ubuntu 22.04
ESP32

I am trying to add custom ros2 messages from a package called openbase2_msgs that contains 2 messages: EncoderData.msg and MotorVels.msg https://github.com/TheHassanShahzad/OpenBase2/tree/master/openbase2_msgs

When i copy paste the folder into /snap/arduino/85/Arduino/libraries/micro_ros_arduino-humble/extras/library_generation/extra_packages and then go back to /snap/arduino/85/Arduino/libraries/micro_ros_arduino-humble/ then run the docker script witj -p set to esp32, firstly i have a permission denied error. but making the .sh script executable fixes that

Then like you, i get lots of packages having an stderr output. For me around 43. When i look inside /src/openbase2_msgs i do see some folders and inside those my 2 .msg files

However when closing Arduino IDE and opening it up again, i cannot even compile an example Publisher for any board. Is my method wrong?

hippo5329, 2024-08-07:
It will be easier to add custom msgs if you switch to platformio. You can add the custom msgs to extra_packages.

echang264, 2025-01-20:
This was resolved for me by changing the `export TOOLCHAIN_PREFIX` line in the teensy4 section of library_generation.sh to:

`export TOOLCHAIN_PREFIX=/uros_ws/teensy-compile/tools/arm/bin/arm-none-eabi-`

There appears to be a [relevant commit with this fix](https://github.com/micro-ROS/micro_ros_arduino/commit/084242c5a6bf90093d60c7ae128575cda0a74e70) that came after the 2.0.7 release.