micro-ROS/micro_ros_arduino #1537: Error building ros_publisher.ino example for Teensy 3.2

tmxkn1, 2023-09-29:
## Issue template

- Hardware description: Teensy 3.2
- RTOS: N/A?
- Installation type: Android IDE 2.2.1
- Version or commit hash: [v2.0.7-humble](https://github.com/micro-ROS/micro_ros_arduino/releases/tag/v2.0.7-humble)

#### Steps to reproduce the issue
I was in general following the tutorial [teensy_with_arduino](https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/). The steps I took differently to the tutorial were:

1. I did not use a docker. Instead, I used my local Ubuntu which has ros2 humble installed.
2. I used appimage version of the Arduino IDE. The hardware folder is under `~/.arduino15/packages/teensy/hardware/avr/1.58.1`. Therefore, in the step that requires to replace the platform.txt, I ran the following:
```
cd ~/.arduino15/packages/teensy/hardware/avr/1.58.1
curl https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/humble/extras/patching_boards/platform_teensy.txt > platform.txt
```
Note that I also used humble instead of foxy in the link.

I followed the tutorial until the last step in "Program the Teensy". The code did not compile.

#### Expected behavior
The code should compile.

#### Actual behavior
I received the following error:

```
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/ice9/Arduino/libraries/micro_ros_arduino-2.0.7-humble/src/mk20dx256
/home/ice9/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ice9/Arduino/libraries/micro_ros_arduino-2.0.7-humble/src/mk20dx256/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xc0): undefined reference to `__locale_ctype_ptr'
/home/ice9/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ice9/Arduino/libraries/micro_ros_arduino-2.0.7-humble/src/mk20dx256/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/ice9/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/ice9/Arduino/libraries/micro_ros_arduino-2.0.7-humble/src/mk20dx256/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x36): undefined reference to `__locale_ctype_ptr'
/home/ice9/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xa6): undefined reference to `__locale_ctype_ptr'
/home/ice9/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x162): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

#### Additional information

tmxkn1, 2023-09-29:
Found a few issues mentioning this error. And within these, I found a temporary solution and it still works:

https://github.com/micro-ROS/micro_ros_arduino/issues/1378#issuecomment-1653329938

I may track down the issue if I get time or no one else is interested in doing so.

pablogs9, 2023-10-02:
This error is related to a version mismatch between the compiler used for building the micro-ROS library and your local compiler. 

Could you provide the compiler version you are using for building your application?

The-Kriz, 2023-10-19:
Arduino: 1.8.5 (Linux), TD: 1.58, Board: "Teensy 4.1, Serial, 600 MHz, Faster, US English"

"/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/arm-none-eabi-g++" -E -CC -x c++ -w  -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=158 -DARDUINO=10805 -DARDUINO_TEENSY41 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-I/home/kriz/arduino-1.8.5/hardware/teensy/avr/cores/teensy4" "-I/home/kriz/Arduino/libraries/micro_ros_arduino/src" "-I/home/kriz/arduino-1.8.5/hardware/teensy/avr/libraries/NativeEthernet/src" "-I/home/kriz/arduino-1.8.5/hardware/teensy/avr/libraries/FNET/src" "/home/kriz/arduino-1.8.5/hardware/teensy/avr/libraries/FNET/src/port/cpu/mimxrt/fnet_mimxrt.c" -o "/dev/null"

/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_643732/sketch/micro-ros_publisher.ino.cpp.o: in function `timer_callback(rcl_timer_s*, long long)':
/home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:36: undefined reference to `rcl_publish'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_643732/sketch/micro-ros_publisher.ino.cpp.o: in function `setup':
/home/kriz/Arduino/libraries/micro_ros_arduino/src/micro_ros_arduino.h:33: undefined reference to `rmw_uros_set_custom_transport'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_643732/sketch/micro-ros_publisher.ino.cpp.o: in function `setup':
/home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:49: undefined reference to `rcutils_get_default_allocator'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:52: undefined reference to `rclc_support_init'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:55: undefined reference to `rclc_node_init_default'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:58: undefined reference to `rclc_publisher_init_default'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:66: undefined reference to `rclc_timer_init_default'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:73: undefined reference to `rclc_executor_init'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:74: undefined reference to `rclc_executor_add_timer'
/home/kriz/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /tmp/arduino_build_643732/sketch/micro-ros_publisher.ino.cpp.o: in function `loop':
/home/kriz/Arduino/libraries/micro_ros_arduino/examples/micro-ros_publisher/micro-ros_publisher.ino:81: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status
Using library micro_ros_arduino at version 2.0.7-humble in folder: /home/kriz/Arduino/libraries/micro_ros_arduino 
Using library NativeEthernet at version 1.0.5 in folder: /home/kriz/arduino-1.8.5/hardware/teensy/avr/libraries/NativeEthernet 
Using library FNET at version 0.1.3 in folder: /home/kriz/arduino-1.8.5/hardware/teensy/avr/libraries/FNET 
Error compiling for board Teensy 4.1.

pablogs9, 2023-10-24:
I cannot see the compiler version here, also this is not the same error as before.