micro-ROS/micro_ros_arduino #1540: Undefined reference to `rmw_uros_discover_agent'

corentin-j, 2023-10-01:
- Hardware description: ESP32
- Installation type: Arduino IDE v2.2.2
- Version or commit hash: humble

### Expected behaviour

The following code with the use of `rmw_uros_discover_agent` function should compile.

### Actual behaviour

I started from the wifi publisher example that worked well. I tryed to implement the discover_agent function but the compilation failed (following error message). I did not find either an exemple that used this function with the arduino IDE.

```Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/corentin/Arduino/libraries/micro_ros_arduino/src/esp32
/home/corentin/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino/sketches/10073CF119A2DB6E50F7D417A9AFB5BB/sketch/micro_ros_publisher_wifi.ino.cpp.o:(.literal._Z5setupv+0x24): undefined reference to `rmw_uros_discover_agent'
/home/corentin/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/arduino/sketches/10073CF119A2DB6E50F7D417A9AFB5BB/sketch/micro_ros_publisher_wifi.ino.cpp.o: in function `setup()':
/home/corentin/Arduino/test_esp/micro_ros_publisher_wifi/micro_ros_publisher_wifi.ino:59: undefined reference to `rmw_uros_discover_agent'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

### My code

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

#include <rmw_microros/discovery.h>

#if !defined(ESP32) && !defined(TARGET_PORTENTA_H7_M7) && !defined(ARDUINO_NANO_RP2040_CONNECT) && !defined(ARDUINO_WIO_TERMINAL)
#error This example is only available for Arduino Portenta, Arduino Nano RP2040 Connect, ESP32 Dev module and Wio Terminal
#endif

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

void setup() {
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  delay(2000);

  allocator = rcl_get_default_allocator();

  rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
  RCCHECK(rcl_init_options_init(&init_options, allocator));
  rmw_init_options_t* rmw_options = rcl_init_options_get_rmw_init_options(&init_options);

  // Auto discover micro-ROS agent
  if(rmw_uros_discover_agent(rmw_options) != RCL_RET_OK){
  printf("micro-ROS agent not found\n");
  }

  // create init_options
  RCCHECK(rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_wifi_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_best_effort(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "topic_name"));

  msg.data = 0;
}

void loop() {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
    delay(500);
}
```

pablogs9, 2023-10-02:
I guess that it is not built because the transport is custom instead of the default UDP or TCP: 

https://github.com/micro-ROS/rmw_microxrcedds/blob/bc4eb312ac4601a4137c35f4a56b9b83b4b18339/rmw_microxrcedds_c/CMakeLists.txt#L231C19-L231C42

Could you test that enabling this flag, it is built when rebuilding the micro-ROS library?

corentin-j, 2023-10-02:
Thanks a lot for the reply ! As I am new to Docker, I will try to compile it first without adding the parameter and then see if it helps

pablogs9, 2023-10-03:
You need to run this command in the project folder

corentin-j, 2023-10-05:
I finally managed to compile it. Is this indicated where the created binaries are located ?

pablogs9, 2023-10-06:
Could you provide instructions on how did you build it?

Binaries are created inside src/[your target] folder, you can check it by using git diff.

corentin-j, 2023-10-07:
I went in my `micro_ros_arduino` folder and executed :
```
docker run -it --rm -v $(pwd):/project --env MICROROS_LIBRARY_FOLDER=extras microros/micro_ros_static_library_builder:humble -p esp32
```
Here is the result of git diff :
```
diff --git a/extras/library_generation/colcon.meta b/extras/library_generation/colcon.meta
index 5bb8eff..a5ed249 100755
--- a/extras/library_generation/colcon.meta
+++ b/extras/library_generation/colcon.meta
@@ -46,7 +46,8 @@
                 "-DRMW_UXRCE_MAX_SERVICES=1",
                 "-DRMW_UXRCE_MAX_CLIENTS=1",
                 "-DRMW_UXRCE_MAX_HISTORY=4",
-                "-DRMW_UXRCE_TRANSPORT=custom"
+                "-DRMW_UXRCE_TRANSPORT=custom",
+                "-DRMW_UXRCE_TRANSPORT_UDP=ON"
             ]
         }
     }
diff --git a/src/action_msgs/msg/detail/goal_info__functions.h b/src/action_msgs/msg/detail/goal_info__functions.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_info__rosidl_typesupport_introspection_c.h b/src/action_msgs/msg/detail/goal_info__rosidl_typesupport_introspection_c.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_info__rosidl_typesupport_microxrcedds_c.h b/src/action_msgs/msg/detail/goal_info__rosidl_typesupport_microxrcedds_c.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_info__struct.h b/src/action_msgs/msg/detail/goal_info__struct.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_info__type_support.h b/src/action_msgs/msg/detail/goal_info__type_support.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_status__functions.h b/src/action_msgs/msg/detail/goal_status__functions.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_status__rosidl_typesupport_introspection_c.h b/src/action_msgs/msg/detail/goal_status__rosidl_typesupport_introspection_c.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_status__rosidl_typesupport_microxrcedds_c.h b/src/action_msgs/msg/detail/goal_status__rosidl_typesupport_microxrcedds_c.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_status__struct.h b/src/action_msgs/msg/detail/goal_status__struct.h
old mode 100755
new mode 100644
diff --git a/src/action_msgs/msg/detail/goal_status__type_support.h b/src/action_msgs/msg/detail/goal_status__type_support.h
old mode 100755
new mode 100644
```

pablogs9, 2023-10-09:
Sorry but with the current information, I cannot determine the problem.

The [command](https://github.com/micro-ROS/docker/blob/iron/micro-ROS-static-library-builder/Dockerfile) you are using is the very same that is [running in our CI](https://github.com/micro-ROS/micro_ros_arduino/blob/iron/.github/workflows/library_generation.yml).

Please report if you find the problem.

corentin-j, 2023-10-12:
I think I need to change several things on the colcon.meta :

```
        "microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=ON",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=ON",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=OFF"
            ]
        },
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=udp"
            ]
        }
```
The issue is that now it does not compile anymore :(. Did you manage to compile it for UDP transport ? 
Here is my compilation error :
```
Finished <<< rcutils [5.76s]
Starting >>> rosidl_runtime_c
Finished <<< tracetools [2.74s]                                                                                                                                   
Starting >>> rcl_logging_interface
--- stderr: microxrcedds_client                                                                                                               
In file included from /uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/transport.h:32,
                 from /uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/util/ping.h:28,
                 from /uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/src/c/util/ping.c:1:
/uros_ws/firmware/mcu_ws/eProsima/Micro-XRCE-DDS-Client/include/uxr/client/profile/transport/ip/udp/udp_transport.h:37:27: error: field 'platform' has incomplete type
     struct uxrUDPPlatform platform;
                           ^~~~~~~~
gmake[2]: *** [CMakeFiles/microxrcedds_client.dir/build.make:289: CMakeFiles/microxrcedds_client.dir/src/c/util/ping.c.obj] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:86: CMakeFiles/microxrcedds_client.dir/all] Error 2
gmake: *** [Makefile:139: all] Error 2
---
Failed   <<< microxrcedds_client [3.39s, exited with code 2]
Aborted  <<< rosidl_parser [2.84s]                                                                                                                 
Aborted  <<< rosidl_runtime_c [3.13s]                                                                                  
Aborted  <<< rcl_logging_interface [1.72s]                             

Summary: 7 packages finished [9.75s]
  1 package failed: microxrcedds_client
  3 packages aborted: rcl_logging_interface rosidl_parser rosidl_runtime_c
  5 packages had stderr output: microxrcedds_client rcl_logging_interface rcutils rosidl_cli rosidl_runtime_c
  58 packages not processed
[ros2run]: Process exited with failure 2
find: ‘firmware/build/include/’: No such file or directory
cp: cannot stat 'firmware/build/include/*': No such file or directory
cp: cannot stat 'firmware/build/libmicroros.a': No such file or directory
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
rsync is already the newest version (3.2.7-0ubuntu0.22.04.2).
0 upgraded, 0 newly installed, 0 to remove and 239 not upgraded.

```

pablogs9, 2023-10-13:
You need to use custom transport in Arduino, it won't compile the Posix UDP transports.

https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/custom_transports/custom_transports.html

corentin-j, 2023-10-13:
Ok thanks a lot ! I think I will use custom transport and forget about discovery for Arduino

closed 2023-10-13 by corentin-j