micro-ROS/micro_ros_arduino #1551: Question about timeout parameter of rmw_uros_ping_agent function

ChoiYouJung, 2023-10-13:
## Issue template

- Hardware description:  Teensy4.1 or i.MX-RT1060 MCU
- RTOS:  No
- Installation type:  Arduino IDE or MCUXpressoIDE
- Version or commit hash:  humble

#### Steps to reproduce the issue

(1) Code Behavior Description
* micro-ros_reconnection_example example code
(examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino)
https://github.com/micro-ROS/micro_ros_arduino/blob/humble/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino

* Question about the Timeout parameter when using the rmw_uros_ping_agent function in the following example code:

**Q1 :  While testing the program, it seems that when using the rmw_uros_ping_agent function, a delay occurs as much as the number in the first parameter, the timeout parameter. Is my understanding correct?**

**Q2 : When timeout is set to 100 in the rmw_uros_ping_agent function, when ping with the agent is successful faster than 100msec, what is the delay of 100msec? Or is it possible to escape the function faster than that?**

```
  switch (state) {
    case WAITING_AGENT:
      EXECUTE_EVERY_N_MS(500, state = (RMW_RET_OK == rmw_uros_ping_agent(100, 1)) ? AGENT_AVAILABLE : WAITING_AGENT;);
      break;
    case AGENT_AVAILABLE:
      state = (true == create_entities()) ? AGENT_CONNECTED : WAITING_AGENT;
      if (state == WAITING_AGENT) {
        destroy_entities();
      };
      break;
    case AGENT_CONNECTED:
      EXECUTE_EVERY_N_MS(200, state = (RMW_RET_OK == rmw_uros_ping_agent(100, 1)) ? AGENT_CONNECTED : AGENT_DISCONNECTED;);
      if (state == AGENT_CONNECTED) {
        rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100));
      }
      break;
    case AGENT_DISCONNECTED:
      destroy_entities();
      state = WAITING_AGENT;
      break;
    default:
      break;
  }
```

**Additional question: If the function set in the second question is implemented so as not to escape quickly, is there a way to reduce the delay?**

#### Expected behavior
* To operate the micro-ros_reconnection_example example code.

* To understand the rmw_uros_ping_agent function in the micro-ros_reconnection_example example code and operate it to minimize delay.

#### Actual behavior
* Running the micro-ros_reconnection_example example code

#### Additional information
* The operating system of the Arduino IDE is Windows 10.
* The operating system of the MCUXpresso IDE is Windows 10.
* Similar to (#1396), I want to use the reconnect function while minimizing latency.

pablogs9, 2023-10-13:
> Q1 : While testing the program, it seems that when using the rmw_uros_ping_agent function, a delay occurs as much as the number in the first parameter, the timeout parameter. Is my understanding correct?

The first parameter is the maximum time that the ping can take. So for example if there is no agent found, this call will take take this number of milliseconds.

> Q2 : When timeout is set to 100 in the rmw_uros_ping_agent function, when ping with the agent is successful faster than 100msec, what is the delay of 100msec? Or is it possible to escape the function faster than that?

Yes, the function will return as fast as the agent responds to the ping. The 100 ms is the maximum amount of time allowed for the ping operation.

> Additional question: If the function set in the second question is implemented so as not to escape quickly, is there a way to reduce the delay?

Just reduce the timeout in order to provide a smaller time window to perform the ping operation.

ChoiYouJung, 2023-10-17:
thank you. Thanks, it was helpful.
I have an additional question.
In the reconnection example, if EXECUTE_EVERY_N_MS(200, state = (RMW_RET_OK == rmw_uros_ping_agent(100, 1)) ? AGENT_CONNECTED : AGENT_DISCONNECTED;); is set in the EXECUTE_EVERY_N_MS function,
**Total maximum waiting time is 200msec, right? Or is 100msec correct?**

pablogs9, 2023-10-17:
Every 200 ms it will try to ping the agent for a maximum time of 100 ms.

ChoiYouJung, 2023-10-17:
oh! I misunderstood.

Thanks for your help.

Close the issue.

closed 2023-10-17 by ChoiYouJung