micro-ROS/micro_ros_arduino #1554: The session has been established, but the nodes are not functioning correctly.

Chuspi29, 2023-10-29:
rpi4master@rpi4master:~$ sudo micro-xrce-dds-agent serial --dev /dev/ttyACM0 -b 115200
[1698546806.114685] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1698546806.115410] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1698546806.554954] info     | Root.cpp           | create_client            | create                 | client_key: 0x25E30594, session_id: 0x81
[1698546806.555178] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x635635092, address: 0

-Im using a raspberry pi pico rp2040, a raspberry pi 4 4gb and a computer to access via ssh to the raspberry pi
- RTOS: FREERTOS
- Installation type: I use a docker to acces micro ros: https://hub.docker.com/r/microros/micro-ros-agent  (Humble)
Also install the micro-xrce-dds-agent following the steps right here: https://snapcraft.io/micro-xrce-dds-agent
- Im using micro ros to connect with ROS2 Humble.

#### Steps to reproduce the issue
Install ROS2 humble using the debian packages.
Follow the instructions that is shown in this pages: 
https://hub.docker.com/r/microros/micro-ros-agent (humble)
https://snapcraft.io/micro-xrce-dds-agent

#### Expected behavior
[1698548607.973741] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1698548608.521272] info     | Root.cpp           | delete_client            | delete                 | client_key: 0x14642262
[1698548608.521385] info     | SessionManager.hpp | destroy_session          | session closed         | client_key: 0x14642262, address: 0
[1698548608.521434] info     | Root.cpp           | create_client            | create                 | client_key: 0x4F4067EA, session_id: 0x81
[1698548608.521459] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x4F4067EA, address: 0
[1698548608.533004] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x4F4067EA, participant_id: 0x000(1)
[1698548608.536969] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x4F4067EA, topic_id: 0x000(2), participant_id: 0x000(1)
[1698548608.539250] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x4F4067EA, publisher_id: 0x000(3), participant_id: 0x000(1)
[1698548608.542347] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x4F4067EA, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1698548608.545870] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x4F4067EA, topic_id: 0x001(2), participant_id: 0x000(1)
[1698548608.548010] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x4F4067EA, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1698548608.552643] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x4F4067EA, datareader_id: 0x000(6), subscriber_id: 0x000(4)

#### Actual behavior
[1698546806.114685] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1698546806.115410] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1698546806.554954] info     | Root.cpp           | create_client            | create                 | client_key: 0x25E30594, session_id: 0x81
[1698546806.555178] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x635635092, address: 0

The expected behavior was obtained by the computer that has the "micro-ros-agent." When you plug in the Raspberry Pi Pico, everything works fine. However, in the case where we are using a Raspberry Pi 4, we can't install the "micro-ros-agent" because the terminal says that it only works on AMD64 systems, not on ARM64. Therefore, we searched for information and found an alternative called "micro-xrce-dds-agent."

With this latter agent, the connection is initialized, but the nodes don't work
[image]
[image]

Chuspi29, 2023-10-29:
This is the code we are using:
#include <iostream>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>
#include <rmw_microros/rmw_microros.h>
#include "FreeRTOS.h"
#include "task.h"
#include "pico/stdlib.h"
#include "hardware/pwm.h"
#include "hardware/adc.h"
#include "hardware/gpio.h"
extern "C" {
    #include "pico_uart_transports.h" // Archivo de encabezado correspondiente
}
#include <std_msgs/msg/float32.h>
#include <geometry_msgs/msg/vector3.h>

volatile long encoder_count = 0;
volatile long encoder_count2 = 0;
//Definicion global pin Led deafultc
#define LED_PIN PICO_DEFAULT_LED_PIN

//Defino pi para posterior uso
#define PI 3.14159265358979323846

float w_motor1=0;
float w_motor2=0;
float U1=0;
float U2=0;    
float e[2]={0,0};
int led_value=0;

float Wi[2] = {13,13};

//---------------------------------------------------------------------             Clase Motor                ----------------------------------------------------------------------------------------
class Motor {
    private : //atributos
    int pin_num_pwm, pin_num_in1, pin_num_in2,sensorA_pin, sensorB_pin, freq=1000 , FC1 , FC2;
    //parametros
    int ppr;
    int decoder_number;
    int gear_ratio;

    absolute_time_t prevT = get_absolute_time();
    double posPrev;
    int slice_num;
    int CH;
    
    public://metodos
    

        //Metodo constructor
        Motor(int _pin_num_pwm, int _pin_num_in1, int _pin_num_in2, int _sensorA_pin, int _sensorB_pin, int _freq){

        this-> pin_num_pwm =    _pin_num_pwm;
        this-> pin_num_in1 =    _pin_num_in1;
        this-> pin_num_in2 =    _pin_num_in2;
        this-> sensorA_pin =    _sensorA_pin;
        this-> sensorB_pin =    _sensorB_pin;
        this-> freq        =    _freq;
                
        gpio_init(this->pin_num_pwm);
        gpio_init(this->pin_num_in1);
        gpio_init(this->pin_num_in2);
        gpio_init(this->sensorA_pin);
        gpio_init(this->sensorB_pin);
[... 321 lines cut ...]
        &publisher_1,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs,msg,Vector3),
        "/reptil321");

    rcl_ret_t rc = rclc_subscription_init_default(
     &subscriber, 
     &node,
     ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Vector3)
     , "/rex123");

    rclc_executor_init(&executor, &support.context, 2, &allocator);
    gpio_put(LED_PIN, 1);

    rclc_executor_add_subscription(
    &executor, &subscriber, &msg,
    &subscription_callback, ON_NEW_DATA);

    xTaskCreate(Print_Task, "Imprimir", 512, NULL, 1, NULL);
    xTaskCreate(stateMachine, "Maquina de estados", 1024, NULL, 2, NULL);
    xTaskCreate(Spin_Task, "Spin", 512, NULL, 4, NULL);
    vTaskStartScheduler();

    return 0;
}

pablogs9, 2023-10-30:
Does the [provided example](https://github.com/micro-ROS/micro_ros_arduino/blob/iron/examples/micro-ros_publisher/micro-ros_publisher.ino) works as expected?