micro-ROS/micro_ros_arduino #1559: Unable to read the string from the topic

ManoharaDraksharam, 2023-11-03:
## Issue template

- Hardware description: 
- RTOS: 
- Installation type: 
- Version or commit hash: 

Hello in the following code i am trying to read the string from ROS topic and convert the string to integer, but i was unable to read the string from the topic. can someone help me the mistake was.

Thanks 

#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/string.h>
#include <ESP32Servo.h>

rcl_subscription_t subscriber;
std_msgs__msg__String msg;
rclc_executor_t executor;
rcl_allocator_t allocator;
rclc_support_t support;
rcl_node_t node;

Servo myservo1;
Servo myservo2;
Servo myservo3;
Servo myservo4;

int pos_servo1;   
int pos_servo2;
int pos_servo3;
int pos_servo4; 

int pos_rec1;
int pos_rec2;
int pos_rec3;
int pos_rec4;

char char_sep = ',';

const int str_array_size = 4;

String received_arr[str_array_size]={};

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

void error_loop(){
  while(1){
    // digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void subscription_callback(const void *msgin) 
{
  const std_msgs__msg__String * msg = (const std_msgs__msg__String *)msgin;

  String a = String(msg->data.data);

  char b[15];
  a.toCharArray(b,15);
  // strcpy(b, msg->data.data);
  String received = "";
  int count =0; 
  //a.toCharArray(b,15);

  for (int i=0; i < (sizeof(b))-1; i++)
  {
    if (b[i] == char_sep ) 
    {
      received_arr[count]=received;
      count +=1;
      received ="";
    }
    else
    { 
      received += b[i];
    }        
  }        

  pos_rec1=received_arr[0].toInt();
  pos_rec2=received_arr[1].toInt();
  pos_rec3=received_arr[2].toInt();
  pos_rec4=received_arr[3].toInt();

  myservo1.write(pos_rec1);
  delay(1500);
  myservo2.write(pos_rec2);
  delay(1500);
  myservo3.write(pos_rec3);
  delay(1500);
  myservo4.write(pos_rec4);
  delay(1500);
  myservo4.write(pos_servo4);
  delay(1500);
  myservo3.write(pos_servo3);
  delay(1500);
  myservo2.write(pos_servo2);
  delay(1500);
  myservo1.write(pos_servo1);

  delay(2000);

}

void setup() {

  set_microros_transports();
  pos_servo1 = 0;   
  delay(1000);
  pos_servo2 = 30;
  delay(1000);
  pos_servo3 = 180;
  delay(1000);
  pos_servo4 = 0;

  myservo1.attach(12);
  myservo2.attach(13);
  myservo3.attach(14);
  myservo4.attach(15);
    
  myservo1.write(pos_servo1);  
  myservo2.write(pos_servo2); 
  myservo3.write(pos_servo3); 
  myservo4.write(pos_servo4);
  
  delay(2000);

  allocator = rcl_get_default_allocator();

   //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String),
    "/chatter"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));

}

void loop() {
  delay(1);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

pablogs9, 2023-11-03:
You are not initializing reception string memory, please read this: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#message-memory

You need to provide some memory to `std_msgs__msg__String msg;`

ManoharaDraksharam, 2023-11-04:
I had used utilities to read the string from the topic, and was successful.
i am trying to read the same data for published , but was unable to publish it
```
micro_ros_utilities_create_message_memory(
      ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String),
      &msg_sub,
      (micro_ros_utilities_memory_conf_t) {}
    );
```

```
micro_ros_utilities_create_message_memory(
      ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String),
      &msg_pub,
      (micro_ros_utilities_memory_conf_t) {}
    );
void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {

    RCSOFTCHECK(rcl_publish(&publisher, &msg_pub, NULL));
  
    msg_pub.data = micro_ros_string_utilities_set(msg_pub.data, msg_sub.data);
   
  }
}
```

pablogs9, 2023-11-06:
You are publishing before setting the data

Be careful with I allocating memory inside a timer callback.

closed 2023-11-06 by pablogs9

ManoharaDraksharam, 2023-11-06:
Thanks for the help, it works fine now.