micro-ROS/micro_ros_arduino #1574: Session Not Established

bobkim938, 2023-11-16:
## Issue template

- Hardware description: Arduino Nano rp2040 connect
- RTOS: N/A
- Installation type: Android IDE 2.2.1
- Version or commit hash: v2.0.7 Humble

I'm very new to ROS, so it would be very appreciated if I can figure out the issue.
I wanted to test using the publish example code if I can publish from arduino to my PC. 
I followed the official installation procedures, and I did not make any changes to ROS_DOMAIN_ID.
`ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0 -v6`
The session was not established using the above command.

#### Expected behavior
I expected the session to be established
#### Actual behavior
[image]

pablogs9, 2023-11-16:
Are you resetting the board after instantiating the micro-ROS Agent?

This example do not handle client-agent reconnection, see this for implementing it: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html

bobkim938, 2023-11-16:
Should I reset the board, after instantiating?

pablogs9, 2023-11-16:
Please read: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/handle_reconnections/handle_reconnections.html

bobkim938, 2023-11-16:
Thank you so much, I was able to somehow solve the issue.

closed 2023-11-16 by pablogs9