micro-ROS/micro_ros_arduino #1596: teensy 4.1 parameter server limits to 5 parameters

adityakamath, 2023-12-02:
I have a working implementation of the parameter server with 4 parameters, and I was trying to add two more. However, when I try with 6 parameters, the client goes into the RCCHECK error loop. I removed one of them and tried with 5 parameters, and it works as expected. 

This is my parameter server config:

```
rclc_parameter_options_t param_options = {
        .notify_changed_over_dds = true,
        .max_params = 6,
        .allow_undeclared_parameters = false,
        .low_mem_mode = true};
```

I tried increasing max_params to 10 (while only adding 6 parameters), tried with low_mem_mode = false, but none of these combinations work for more than 5 parameters. Works as expected with 5 or fewer than 5.

Is this limit of 5 hardcoded anywhere? How do I work with more than 5 parameters? Eventually I want to add a few more, up to 10 parameters.

- Hardware description: Teensy 4.1
- RTOS: -
- Installation type: micro_ros_agent (using micro_ros_setup)
- Version or commit hash:humble

#### Steps to reproduce the issue

#### Expected behavior
micro-ros-agent connects to the device, all entities are successfully created, and I can set/get all the parameters

#### Actual behavior
micro-ros-agent tries to connect to the device, the client goes into the RCCHECK error loop while trying to create the parameter server, while adding parameters to the server or while setting the default values. This is only observed when number of parameters is > 5.

#### Additional information

kjrusscher, 2024-06-27:
The implementation is limited to a set number of publishers, subscribers, nodes, etc. You can adjust this to your needs with this file: https://github.com/micro-ROS/micro_ros_arduino/blob/jazzy/extras/library_generation/colcon.meta

adityakamath, 2024-07-13:
I did adjust it before I opened this issue, but it didn't seem to work. I haven't tried it since as I did not get the time. Not sure if this works now..