micro-ROS/micro_ros_arduino #1612: Isolating Micro-ROS topics

Muthukumar4796, 2023-12-16:
When I add the line 'export ROS_DOMAIN_ID=47' to the bashrc file and then load Micro-ROS, my custom topics are not visible. However, when I comment out the ROS domain ID line, restart the agent, and reload Micro-ROS, my custom topics become visible. I'm stuck and unsure how to properly isolate my Micro-ROS topics without interference from the ROS domain ID and Can anyone provide guidance on how to properly isolate Micro-ROS topics in this setup?

Note:For Isolating the ROS2 topics using ROS_DOMAIN_ID is working fine.

pablogs9, 2023-12-18:
You can configure the Domain ID in the micro-ROS application: https://github.com/micro-ROS/micro-ROS-demos/blob/70f3cbf27ce12dc9e57e5095b4233553e207d866/rclc/configuration_example/configured_publisher/main.c#L49

Muthukumar4796, 2023-12-18:
After configuring my domain ID with Micro-ROS Arduino code, when I call the Micro-ROS agent, it does not display my topics.