micro-ROS/micro_ros_arduino #1614: Unable to compile any example sketches for Teensy boards on Arduino IDE 2

Darkextratoasty, 2023-12-19:
## Issue template

- Hardware description: 
Trying to use a Teensy 4.1, Arduino IDE is installed on Windows 11 PC. 

- Version or commit hash: 
v2.0.7-iron
Arduino IDE 2.2.1

#### Steps to reproduce the issue
- Install Arduino IDE 2.2.1 on a Windows 11 PC with no existing Arduino IDE installation.
- Add line "https://www.pjrc.com/teensy/package_teensy_index.json" to preferences page and install Teensy support as described here: https://www.pjrc.com/teensy/td_download.html
- Set board to any of the Teensy boards (I tried 2 up to 4.1)
- Grab ZIP file of v2.0.7-iron release of micro-ros-arduino from here: https://github.com/micro-ROS/micro_ros_arduino/archive/refs/tags/v2.0.7-iron.zip
- Add library to Arduino using the "Add ZIP Library" function
- Open any example sketch and hit compile

#### Expected behavior
Expected to compile

#### Actual behavior
Exits status 1
The full output from an un-cached compile is in the pastebin here (https://pastebin.com/SZzaMhxC) and the errors from a cached compile are below. In the outputs below, I'm compiling the "micro-ros_publisher" example for the Teensy 4.1.

Error section of cached compile:
```
Compiling core...
Using precompiled core: C:\Users\joshu\AppData\Local\Temp\arduino\cores\teensy_avr_teensy41_usb_serial,speed_600,opt_o2std,keys_en-us_2060ad08b051364d739a8fa53272fcc6\core.a
Linking everything together...
"C:\\Users\\joshu\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax "-TC:\\Users\\joshu\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1\\cores\\teensy4/imxrt1062_t41.ld" -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -o "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D/micro-ros_publisher.ino.elf" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\sketch\\micro-ros_publisher.ino.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\micro_ros_arduino\\default_transport.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\micro_ros_arduino\\native_ethernet_transport.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\micro_ros_arduino\\wifi_transport.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\NativeDns.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\NativeEthernet.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\NativeEthernetClient.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\NativeEthernetServer.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\NativeEthernetUdp.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\NativeMdns.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\NativeEthernet\\Nativesocket.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\fnet_cpu.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_eth.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_isr.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_isr_inst.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_serial.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_serial.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\fnet_usb.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\fnet_usb_config.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\port\\netif\\fec\\fnet_fec.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\autoip\\fnet_autoip.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_lock.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_platform.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_socketio.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_threadapi.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_tickcounter.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_tlsio.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\azure\\fnet_azure_tlsio_socketio.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\bench\\fnet_bench_cln.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\bench\\fnet_bench_srv.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\dhcp\\fnet_dhcp.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\dhcp\\fnet_dhcp_cln.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\dhcp\\fnet_dhcp_srv.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\dns\\fnet_dns.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\flash\\fnet_flash.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\fnet_service.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\fs\\fnet_fs.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\fs\\fnet_fs_rom.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\fs\\fnet_fs_root.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\http\\fnet_http_cln.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\http\\fnet_http_srv.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\http\\fnet_http_srv_auth.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\http\\fnet_http_srv_cgi.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\FNET\\service\\http\\fnet_http_srv_get.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\7B1CBFBD3316DAACF9817234CBE8FC9D\\libraries\\ [... cut ...]
Using library FNET at version 0.1.3 in folder: C:\Users\joshu\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\libraries\FNET 
exit status 1

Compilation error: exit status 1
```

There are a few "undefined reference" errors pointing to things like "rcl_publish" and "rclc_executor_spin_some". I tried googling these errors but I was only able to find two articles that addressed these specific references (neither of which were helpful for my situation) and a bunch of articles about generic "undefined reference" errors.

pablogs9, 2023-12-20:
did you do this: https://github.com/micro-ROS/micro_ros_arduino?tab=readme-ov-file#patch-teensyduino ?

Darkextratoasty, 2023-12-20:
I had not, I saw the Linux only command "export" and assumed it was a Linux only patch. Having now replaced the platform.txt file, I get the same error, just with different references:
(I did restart the Arduino IDE a few times to make sure the patch took effect)
```
Compiling core...
Using precompiled core: C:\Users\joshu\AppData\Local\Temp\arduino\cores\teensy_avr_teensy41_usb_serial,speed_600,opt_o2std,keys_en-us_2060ad08b051364d739a8fa53272fcc6\core.a
Linking everything together...
"C:\\Users\\joshu\\AppData\\Local\\Arduino15\\packages\\teensy\\tools\\teensy-compile\\11.3.1/arm/bin/arm-none-eabi-gcc" -O2 -Wl,--gc-sections,--relax "-TC:\\Users\\joshu\\AppData\\Local\\Arduino15\\packages\\teensy\\hardware\\avr\\1.58.1\\cores\\teensy4/imxrt1062_t41.ld" -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -o "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13/micro-ros_publisher.ino.elf" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\sketch\\micro-ros_publisher.ino.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\micro_ros_arduino\\default_transport.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\micro_ros_arduino\\native_ethernet_transport.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\micro_ros_arduino\\wifi_transport.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\NativeDns.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\NativeEthernet.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\NativeEthernetClient.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\NativeEthernetServer.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\NativeEthernetUdp.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\NativeMdns.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\NativeEthernet\\Nativesocket.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\fnet_cpu.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_eth.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_isr.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_isr_inst.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_serial.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\cpu\\mimxrt\\fnet_mimxrt_serial.cpp.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\fnet_usb.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\fnet_usb_config.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\port\\netif\\fec\\fnet_fec.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\autoip\\fnet_autoip.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_lock.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_platform.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_socketio.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_threadapi.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_tickcounter.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_tlsio.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\azure\\fnet_azure_tlsio_socketio.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\bench\\fnet_bench_cln.c.o" "C:\\Users\\joshu\\AppData\\Local\\Temp\\arduino\\sketches\\BB65D6B2C2B5874E932AF480405DFB13\\libraries\\FNET\\service\\bench\\fnet_bench_srv.c [... cut ...]
collect2.exe: error: ld returned 1 exit status

Using library micro_ros_arduino at version 2.0.7-iron in folder: C:\Users\joshu\Documents\Arduino\libraries\micro_ros_arduino 
Using library NativeEthernet at version 1.0.5 in folder: C:\Users\joshu\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\libraries\NativeEthernet 
Using library FNET at version 0.1.3 in folder: C:\Users\joshu\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.58.1\libraries\FNET 
exit status 1

Compilation error: exit status 1
```

adityakamath, 2023-12-21:
You might have forgotten to close the Teensy Loader when restarting the IDE. I had very similar errors, fixed after closing the Teensy Loader, restarting the Arduino IDE and trying to compile again (the IDE should automatically open the Teensy Loader again)

Darkextratoasty, 2023-12-21:
Compile/Verify doesn't open the teensy loader, it only opens on upload, so it was never open during my tests. I just tried it again after a fresh restart of my PC and it gave the same error as above.

chandravaran, 2024-02-21:
Hi, I am trying to do the same thing, but I don't see an hardware/avr folders in my arduino folder. I am using 2.3.1. I have teensyduino installed and using teensy 4.1. What exactly are the paths I would have to export

srujanpanuganti, 2024-03-18:
@chandravaran , one workaround is to change the teensyduino version to 1.57.x to make it work as explained in  #1634

chuck0New0Mexico, 2024-04-03:
I am still learning and I appreciate any and all help.   I am attempting to run Micro-Ros on 4.1 Teensy with Arduino IDE  running on either Ubuntu or Windows 10.   I first Installed latest Arduino 2.3.2, but had errors (due to Micro-Ros library being precompiled on when attempting to run micro-ros programs associated with micro-ros libraires following the micro-ros tutorial: "https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/".    I then saw that there was a patch for when an older version of Arduino is run (1.8.19) along with Teensyduino.   I have executed the patch which changes the paltform.txt file, but once done so - I can no longer compile ANY sketches on either Windows or Ubuntu due to same error (shown below).   I see a potential work around utilizing an earlier version of Teensyduino (1.57), however, I can only l download and run Teensyduino 1.59 from the PJRC site.   

thanks, Chuck

Arduino: 1.8.19 (Windows 10), TD: 1.59, Board: "Teensy 4.0, Serial, 600 MHz, Faster, US English"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware 

C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools 

C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\User\Documents\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10819 -build-path 

C:\Users\User\AppData\Local\Temp\arduino_build_31259 -warnings=none -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_534175 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries 

C:\Users\User\Documents\Arduino\libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=600,opt=o2std,keys=en-us -ide-version=10819 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_31259 -warnings=none -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_534175 -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino

Using board 'teensy40' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr

Using core 'teensy4' from platform in folder: C:\Program Files (x86)\Arduino\hardware\teensy\avr

Detecting libraries used...
	
/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++17 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=159 -DARDUINO=10819 -DARDUINO_TEENSY40 -DF_CPU=600000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH "-IC:\\Program Files (x86)\\Arduino\\hardware\\teensy\\avr\\cores\\teensy4" 

"C:\\Users\\User\\AppData\\Local\\Temp\\arduino_build_31259\\sketch\\Blink.ino.cpp" -o nul

exec: "/arm/bin/arm-none-eabi-g++": file does not exist

Error compiling for board Teensy 4.0.