micro-ROS/micro_ros_arduino #1615: Questions about the parameters that determine the number of topics and topic size for publishers and subscribers

ChoiYouJung, 2023-12-20:
## Issue template

- Hardware description:  Teensy4.1 or i.MX-RT1060 MCU
- RTOS:  No
- Installation type:  Arduino IDE or MCUXpressoIDE
- Version or commit hash:  humble

#### Steps to reproduce the issue

(1) Code Behavior Description
The difference between the first and second operation codes is that among the types of ROS messages transmitted from the SBC (Linux) to the MCU, one joint_state message (joint_state_2) has been changed from 1000ms to 30msec.

The second operation code had a data loss of up to 10 pieces, but the second operation code had an issue in which several ROS messages were lost out of the total received data.

The operating time of the code ranged from a minimum of 3 hours to a maximum of 10 hours.

[1] Action code first
- Code that operates 16 publishers, 6 subscriptions, and 10 ROS Timers

- Types of Publish ROS Topic and ROS Timer operation cycle
odom(odometry)(30mesc), battery(battery)(500mec), UInt8MultiArray_1(UInt8MultiArray)(100mec), UInt8MultiArray_2(UInt8MultiArray)(50mec), UInt16MultiArray_2(UInt16MultiArray)(100msec), UInt16MultiArray_1(UInt16MultiArray)(500msec), joint_state_1(joint_state)(100msec), imu(sensor_imu)(30msec), version(string)(1000msec), test_message(int32)(1000msec)

- Types of Subscription ROS Topics
twist, bool_1(Bool), bool_2(Bool), ColorRGBA_1(ColorRGBA), ColorRGBA_2(ColorRGBA), joint_state_2(jointstate)

[2] Action code second
- Code that operates 16 publishers, 6 subscriptions, and 10 ROS Timers

- Types of Publish ROS Topic and ROS Timer operation cycle
odom(odometry)(30mesc), battery(battery)(500mec), UInt8MultiArray_1(UInt8MultiArray)(100mec), UInt8MultiArray_2(UInt8MultiArray)(50mec), UInt16MultiArray_2(UInt16MultiArray)(100msec), UInt16MultiArray_1(UInt16MultiArray)(500msec), joint_state_1(joint_state)(100msec), imu(sensor_imu)(30msec), version(string)(1000msec), test_message(int32)(1000msec)

- Types of Subscription ROS Topics
twist, bool_1(Bool), bool_2(Bool), ColorRGBA_1(ColorRGBA), ColorRGBA_2(ColorRGBA), joint_state_2(jointstate)

For reference, the RQT graph is as follows.
[image]

The micro_ros_node in the center is Teensy4.1 or i.MX-RT1060 MCU.

**Question**
Q1.  When the amount of ROS message data received or transmitted increases, as in the second operation code, are there any parameters that can be adjusted?

Q2.  When I modified the following parameters in the colcon.meta parameter, I modified two of the parameters of the RMW Micro XRCE-DDS implementation.
Is my understanding of the following parameters correct? If not, what parameters can be modified to enable the code to respond when the amount of data sent or received increases?

https://github.com/micro-ROS/rmw_microxrcedds

RMW_UXRCE_STREAM_HISTORY_INPUT  
// Data storage buffer when ROS message is received (maximum number of data received)
// Data transmission direction: SBC (linux) -> MCU (Teensy or i.MXRT1060)

RMW_UXRCE_STREAM_HISTORY_OUTPUT
// Data storage buffer when sending a ROS message (maximum number of data transmitted)
// Data transmission direction: SBC (linux) <- MCU (Teensy or i.MXRT1060)

(For reference, I changed RMW_UXRCE_STREAM_HISTORY_INPUT and RMW_UXRCE_STREAM_HISTORY_OUTPUT from the default value of 4 to 8.)

Q3. Additionally, I have a question about managing memory. Are the input buffer and output buffer at the following address the buffers that store the data when the ROS message is transmitted?
And is it correct that the buffer is captured in the stack section of RAM memory?

https://micro.ros.org/docs/concepts/middleware/memo_prof/

#### Expected behavior
- Operates with less data loss

#### Actual behavior
[1] Action code first
- I tried running this code on Teensy4.1 or i.MX-RT1060 MCU.
- Result: almost no data loss.

[2] Action code second
- I ran this code on the i.MX-RT1060 MCU.
- Result: Data loss occurs in some ROS messages.

#### Additional information
- The operating system of the Arduino IDE is Windows 10.
- The operating system of the MCUXpresso IDE is Windows 10.

pablogs9, 2023-12-20:
Hello, your two scenarios are basically the same, which is the difference between `[1] Action code first` and `[2] Action code second`

> [1] Action code first
>
> Code that operates 16 publishers, 6 subscriptions, and 10 ROS Timers
>
> Types of Publish ROS Topic and ROS Timer operation cycle
> odom(odometry)(30mesc), [...]

Regarding your questions:

> Q1. When the amount of ROS message data received or transmitted increases, as in the second operation code, are there any parameters that can be adjusted?

That's it, you can tune the Micro XRCE-DDS layer and the RMW layer to adjust the middleware operation, but more importantly, your micro-ROS application execution shall be correctly designed to handle your pubs and subs. Could you share your application code?

For example, I assume that you are using a serial port for communication micro-ROS with the Agent, assuming 115200 bauds, you will have an approx bandwidth of ~115 kbps or ~14 kB/s, have you analyzed the size and rates of your publications and subscriptions does not exceed this rate?

> Q2. When I modified the following parameters in the colcon.meta parameter, I modified two of the parameters of the RMW Micro XRCE-DDS implementation.
> Is my understanding of the following parameters [...]

Those two parameters are involved in the buffering mechanism of the middleware, so increasing them will help to have a better performance. Remember that the stream histories sizes shall be power of 2.

> Q3. Additionally, I have a question about managing memory. Are the input buffer and output buffer at the following address the buffers that store the data when the ROS message is transmitted?

Yes

> And is it correct that the buffer is captured in the stack section of RAM memory?

By default all memory in micro-ROS client is statically allocated, so normally it exists in .BSS or .DATA sections.

> Result: Data loss occurs in some ROS messages.

BTW, are your publishers and subscribers reliable or best effort?

ChoiYouJung, 2024-01-22:
Sorry for not being able to share the entire application code.

If we simplify the application code and share the structure, it is as follows.

```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>
#include <nav_msgs/msg/odometry.h>
#include <sensor_msgs/msg/battery_state.h>
#include <std_msgs/msg/string.h>
#include <sensor_msgs/msg/joint_state.h>
#include <std_msgs/msg/multi_array_layout.h>
#include <std_msgs/msg/multi_array_dimension.h>
#include <std_msgs/msg/u_int16_multi_array.h>
#include <std_msgs/msg/u_int8_multi_array.h>
#include <sensor_msgs/msg/imu.h>
#include<geometry_msgs/msg/twist.h>
#include <std_msgs/msg/bool.h>
#include <std_msgs/msg/color_rgba.h>

// publisher
rcl_publisher_t test_pub1;
rcl_publisher_t odom_pub;
rcl_publisher_t battery_pub;
rcl_publisher_t joint_state_pub;
rcl_publisher_t cliff_pub;
rcl_publisher_t state1_test_pub;
rcl_publisher_t state2_test_pub;
rcl_publisher_t ir_pub;
rcl_publisher_t test_pub2;
rcl_publisher_t imu_pub;

rcl_publisher_t cmd_vel_echo_pub;
rcl_publisher_t test_cmd1_echo_pub;
rcl_publisher_t test_cmd2_echo_pub;
rcl_publisher_t test_cmd3_echo_pub;
rcl_publisher_t test_cmd4_echo_pub;
rcl_publisher_t cmd_joint_echo_pub;

// ROS message - publisher message
std_msgs__msg__Int32 test_msg1;
nav_msgs__msg__Odometry odom_msg;
sensor_msgs__msg__BatteryState battery_state_msg;
sensor_msgs__msg__JointState joint_state_msg;
std_msgs__msg__MultiArrayDimension cliff_dim[1];
std_msgs__msg__UInt16MultiArray cliff_msg;
std_msgs__msg__MultiArrayDimension state1_dim[1];
std_msgs__msg__UInt16MultiArray state1_msg;
std_msgs__msg__MultiArrayDimension state2_dim[1];
std_msgs__msg__UInt8MultiArray state2_msg;
std_msgs__msg__MultiArrayDimension ir_dim[1];
std_msgs__msg__UInt8MultiArray ir_msg;
std_msgs__msg__String test_msg2;
sensor_msgs__msg__Imu imu_msg;

geometry_msgs__msg__Twist cmd_vel_echo_msg;
std_msgs__msg__Bool test_cmd1_echo_msg;
std_msgs__msg__Bool test_cmd2_echo_msg;
std_msgs__msg__ColorRGBA test_cmd3_echo_msg;
std_msgs__msg__ColorRGBA test_cmd4_echo_msg;
sensor_msgs__msg__JointState cmd_joint_echo_msg;

// subscription
[... 433 lines cut ...]
		case AGENT_CONNECTED:
			EXECUTE_EVERY_N_MS(100, mros_states = (RMW_RET_OK == rmw_uros_ping_agent(5, 2)) ? AGENT_CONNECTED : AGENT_DISCONNECTED;);
			if (mros_states == AGENT_CONNECTED){
				rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
			}
			break;
		case AGENT_DISCONNECTED:
			destroy_entities();
			mros_states = WAITING_AGENT;
			break;
		default:
			break;
	}

}
```

The answers to the following questions are as follows.

Question:
BTW, are your publishers and subscribers reliable or best effort?

Answer:
I know that it is set to reliable if you use the rclc_publisher_init_default or rclc_subscription_init_default function.
Therefore, I set the publisher and subscriber to reliable,