micro-ROS/micro_ros_arduino #1628: "Subscriber error : Deserialization of data failed -> function deserialize_change" error on the output

anasderkaoui, 2024-01-08:
## Issue template

- Hardware description: Arduino Portenta H7
- Version or commit hash: foxy

#### Steps to reproduce the issue

When I compile the code on the arduino IDE and upload it to the Arduino h7 board everything works fine and the publisher and suscriber show up. *(Unlike with the **ros2arduino** package where there is no publisher showing up, I don't know if there is a fix yet or not)*
The next thing is to run the micro_ros agent and then echo the topic !
I get the data I need (distance to obstacle using three ultrasonic sensors), but from time to time I get this error **`Subscriber error : Deserialization of data failed -> function deserialize_change`**

And I don't know why I get it. I looked up everywhere I could but did not find a fix for it ! Any suggestions are welcome !

#### Additional information

[Code.txt](https://github.com/micro-ROS/micro_ros_arduino/files/13861777/Code.txt)

[image]

abrader-capg, 2025-03-12:
I can into the issue by a ros version mismatch

hippo5329, 2025-03-13:
You can not use micro-ros serial transport and serial print at the same time.

anasderkaoui, 2025-03-18:
Thank you for your reply !
I don't have this setup anymore but I may try to reproduce it another time.