micro-ROS/micro_ros_arduino #1634: Error when compiling micro-ros_publisher example

amangan, 2024-01-14:
## Issue template

- Hardware description: Teensy 4.1, Arduino 2.2.1, Teenseyduino 1.58.1
- Installation type: Humble (reinstalled today)
- Version or commit hash: 

#### Steps to reproduce the issue

1. Install Humble
2. Follow micro ros installation steps
3. Install Arduino 2.2.1 and Teensyduino 1.58.1
4. Follow "Patch Teensyduino" (tried and "Patch Sam" option as well)  -- under ` /.arduino15/packages/teensy/hardware/avr/1.58.1 `
5. Download zip and add folder .zip folder from https://github.com/micro-ROS/micro_ros_arduino/releases
6.  Compile on Arduino IDE

#### Expected behavior

Expecting data writing in example shown (https://micro.ros.org/docs/tutorials/core/teensy_with_arduino/)

#### Actual behavior
Two things: mainly believe #1 is the core issue

1.Compiling the micro-ros_publisher example on the Arduino IDE and getting the following error:

```
WARNING: library micro_ros_arduino claims to run on stm32, OpenCR, Teensyduino, samd, sam, mbed, esp32, mbed_portenta architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
Library micro_ros_arduino has been declared precompiled:
Using precompiled library in /home/amangan/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard
/home/amangan/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/amangan/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_full_topic_name.c.obj): in function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xbe): undefined reference to `__locale_ctype_ptr'
/home/amangan/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/amangan/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librmw-validate_node_name.c.obj): in function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x5e): undefined reference to `__locale_ctype_ptr'
/home/amangan/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /home/amangan/Arduino/libraries/micro_ros_arduino/src/imxrt1062/fpv5-d16-hard/libmicroros.a(librcl-validate_topic_name.c.obj): in function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x46): undefined reference to `__locale_ctype_ptr'
/home/amangan/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0xb6): undefined reference to `__locale_ctype_ptr'
/home/amangan/.arduino15/packages/teensy/tools/teensy-compile/11.3.1/arm/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: validate_topic_name.c:(.text.rcl_validate_topic_name+0x164): undefined reference to `__locale_ctype_ptr'
collect2: error: ld returned 1 exit status
```

2. I am able to run the docker, but it does not get past logger setup:

amangan@amangan-XPS-15-7590:~/microros_ws$ sudo docker run -it --rm -v /dev:/dev --privileged --net=host microros/micro-ros-agent:humble serial --dev /dev/ttyACM0 -v6
[1705205523.023869] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1705205523.024098] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 6

#### Additional information

pablogs9, 2024-01-15:
This is a known issue related to compiler version mismatch. Could you provide your compiler version?

amangan, 2024-01-15:
Is it a mismatch with the downloaded mirco-ros example folder? That may make sense as I am able to compile/upload my own project and _Teensy example (Blink) to the board. Checking through my terminal, it looks like its 11.4.0

```
amangan@amangan-XPS-15-7590:~$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)
```

jhiggason, 2024-01-16:
I have this same issue.  Same version of GCC 11.4.0 on Ubuntu 22.04 Trying to get humble + micro-ros to flash on a Teensy 4.0

pablogs9, 2024-01-16:
Hello @jhiggason, the way will be modifying the docker to include the required version of GCC: https://github.com/micro-ROS/docker/blob/iron/micro-ROS-static-library-builder/Dockerfile

And then modify the building script to use the proper version: https://github.com/micro-ROS/micro_ros_arduino/blob/3455f98c5bb6c7137c0a7e9449ce57b5d68697df/extras/library_generation/library_generation.sh#L167

Let me know if you find a solution and if so, PRs and contributions are welcome.

jhiggason, 2024-01-16:
I found a workaround for the time being.  

Windows 10
Arduino 2.2.1
Teensyduo 1.57.2 (Downgraded from 1.58.1)
Using this Platform.txt > https://raw.githubusercontent.com/micro-ROS/micro_ros_arduino/iron/extras/patching_boards/platform_teensy.txt  

Question though- I don't see any reference to ROS2 Iron in the platform.txt....if I'm using humble- will that be a problem?  Going to test here today at some point but hoping not.

PaulStoffregen, 2024-01-26:
I can confirm Teensyduino 1.58 switched to gcc 11.3.  Prior versions were using gcc 5.4.

Version 1.59 (coming soon and available now as beta with version 0.59.4 and 0.59.5 in Arduino IDE 2.2.x Boards Manager) will begin compiling with C++17 dialect.  Prior versions used C++14 or C++11.

pablogs9, 2024-01-29:
> I can confirm Teensyduino 1.58 switched to gcc 11.3. Prior versions were using gcc 5.4.

Thanks @PaulStoffregen. 

That means that we shall update the compiler version used for building the micro-ROS precompiled library.

amangan, 2024-01-29:
Thank you all for the responses. I am not as familiar with docker, and can try to modify the docker to include the required version of GCC (Teensyduino 1.58.2 and gcc 11.3/11.4)

Or if you recommend, I can wait for Teensyduino 1.59 and the updated micro-ROS precompiled library. Is there a date when those might happen? Not trying to be pushy, I am newer to the open-source SW community and not sure of the norms

srujanpanuganti, 2024-03-18:
@jhiggason , that workaround is sweet!. Also, the issue persists in Teensyduino 1.59.0 too.