micro-ROS/micro_ros_arduino #1651: ESP32 compiler warning in wifi_transport.cpp

kaiaai, 2024-01-27:
## Issue template

- Hardware description: 
Arduino ESP32
- RTOS: 
ESP32
- Installation type: 
Arduino library
- Version or commit hash: 
iron

#### Steps to reproduce the issue

- enable warnings in Arduino IDE
- build any code that includes Arduino micro-ROS

#### Expected behavior
No warning

#### Actual behavior
```
C:\Users\user\Arduino\libraries\micro_ros_kaia\src\wifi_transport.cpp: In function 'size_t arduino_wifi_transport_read(uxrCustomTransport*, uint8_t*, size_t, int, uint8_t*)':
C:\Users\ilya\Arduino\libraries\micro_ros_kaia\src\wifi_transport.cpp:61:20: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
     return (readed < 0) ? 0 : readed;
             ~~~~~~~^~~
```

#### Additional information

pablogs9, 2024-01-29:
Could you check if this solves the issue: https://github.com/micro-ROS/micro_ros_arduino/pull/1653 ?

kaiaai, 2024-01-31:
Thank you - this does remove the warning and solve the issue #1653.

pablogs9, 2024-02-01:
https://github.com/micro-ROS/micro_ros_arduino/pull/1653 closed, solving this

closed 2024-02-01 by pablogs9