micro-ROS/micro_ros_arduino #1660: ESP32 Limited to 10Hz publish and subscribe

ErikParkerrr, 2024-02-01:
## Issue template

- Hardware description: ESP32-WROOM @240Mhz
- RTOS: I'm just using the arduino library
- Installation type: Arduino IDE default everything 
- Version or commit hash: Arduino Humble

#### Steps to reproduce the issue
Modify the example micro-ros_publisher.ino to have a `timer_timeout` to be 10 and load on to your ESP32.
 ros2 topic hz /micro_ros_arduino_node_publisher reveals around 9.7Hz.
Now, do the same thing with `timer_timeout` of 8. The rate plummets to under 5Hz. 

On the subscription side of things, I've noticed that if I publish any faster than 10Hz, a buffer seems to fill up. Latency starts to grow. For example, if I control the on board LED with a topic, I can publish a 1 and zero rapidly and then kill that node. It will continue to flash for a bit as it seems to work its way through the buffer.

I tried this both over wifi and serial.

    docker run -it --net=host microros/micro-ros-agent:humble udp4 -p 8888

    sudo docker run -it --rm -v /dev:/dev -v /dev/shm:/dev/shm --privileged --net=host --pid=host microros/micro-ros-agent:humble serial --dev /dev/ttyUSB0 -v6

exact same results.

#### Expected behavior

Significantly higher rates than 10Hz for both subscribing and publishing.

#### Actual behavior

Very low publishing and subscription rates

#### Additional information

None atm

pablogs9, 2024-02-02:
Did you remove the `delay(100);` from the loop function?

ErikParkerrr, 2024-02-03:
I feel like such a fool right now. I removed that earlier, but also removed the line below it by accident. It didn't connect to the agent properly so I just assumed it had to be there. 

I removed just the delay and I'm getting over 100Hz. So much better.

Thanks for pointing out my silliness and thank you so much for your awesome work

closed 2024-02-03 by ErikParkerrr

pablogs9, 2024-02-03:
Good to hear those 100 Hz! Enjoy micro-ROS