micro-ROS/micro_ros_arduino #1716: Laserscan message Best Effort QOS not working

Tiim12312, 2024-03-28:
Using esp32 dev module with Ros2 iron. 

#### Steps to reproduce the issue
I can create laserscan publishers no problem and correctly publish messages. However this only works with QOS reliable settings. If I only change to best effort QOS it will compile and run but nothing gets published. Again I change nothing else on this correctly working published other then the QOS. I can make best effort QOS publishers for other message types no problem, it's only laserscan that is the issue. 

#### Expected behavior
Should be able to use best effort with laserscan like other message types.

Tiim12312, 2024-04-01:
Ah that makes complete sense, so if I was using UDP would I set it in this section of the cmake like this?

"microxrcedds_client": {
            "cmake-args": [
                "-DUCLIENT_PIC=OFF",
                "-DUCLIENT_PROFILE_UDP=OFF",
                "-DUCLIENT_PROFILE_TCP=OFF",
                "-DUCLIENT_PROFILE_DISCOVERY=OFF",
                "-DUCLIENT_PROFILE_SERIAL=OFF",
                "-UCLIENT_PROFILE_STREAM_FRAMING=ON",
                "-DUCLIENT_PROFILE_CUSTOM_TRANSPORT=ON"
		"-UCLIENT_UDP_TRANSPORT_MTU=1024"
            ]
        },

Or should it go in this section?

"rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=1",
                "-DRMW_UXRCE_MAX_CLIENTS=1",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }

I apologize, I don't know the structure of the system fully yet. Thank you!!

pablogs9, 2024-04-02:
It is explained here: https://docs.vulcanexus.org/en/latest/rst/tutorials/micro/memory_management/memory_management.html#id1