micro-ROS/micro_ros_arduino #1742: Trouble running a stepper motor with micro-ros

Marcal8, 2024-05-01:
I'm having some trouble trying to move a stepper motor with my teensy 4.1. I'm using ROS2 Iron on my Ubuntu 22.04 host computer and micro-ros Iron for the communication with the teensy.

I'm trying to move a stepper using the AccelStepper library and micro-ros. I want to send an especific position to the teensy and have the stepper move to that position. The issue I am having is with the function to call the subscriber, it stops the code for a few microseconds and forbids the necessary steps to be send to move the stepper. This results in the stepper moving in a very slow velocity.

The code I'm using is the following:
    
    #include <AccelStepper.h>
    
    #include <micro_ros_arduino.h>
    
    #include <stdio.h>
    #include <rcl/rcl.h>
    #include <rcl/error_handling.h>
    #include <rclc/rclc.h>
    #include <rclc/executor.h>
    
    #include <geometry_msgs/msg/quaternion.h>
    
    rcl_subscription_t subscriber;
    geometry_msgs__msg__Quaternion msg;
    rclc_executor_t executor;
    rclc_support_t support;
    rcl_allocator_t allocator;
    rcl_node_t node;
    rcl_timer_t timer;
    
    
    const int numSteppers = 1;
    
    const int stepperPullPins[numSteppers] = {7};
    const int stepperDirPins[numSteppers] = {8};
    const int stepperEnablePins[numSteppers] = {9};
    
    AccelStepper stepper[numSteppers] {
      AccelStepper(1,stepperPullPins[0],stepperDirPins[0])
    };
    
    
    float steps = 8000.0;
    
    int posObjectiu[numSteppers] = {0};
    
    
    #define LED_PIN 13
    
    #define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
    #define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}
    
    void error_loop(){
      while(1){
        digitalWrite(LED_PIN, !digitalRead(LED_PIN));
        delay(100);
      }
    }
    
    void subscription_callback(const void * msgin)
    { 
      const geometry_msgs__msg__Quaternion * msg = (const geometry_msgs__msg__Quaternion *)msgin;
    
      posObjectiu[0]=int(msg->w);
    }
    
    
    void setup() {
      set_microros_transports();
     
      pinMode(LED_PIN, OUTPUT);
      digitalWrite(LED_PIN, HIGH); 
     
      delay(2000);
    
      allocator = rcl_get_default_allocator();
    
      //create init_options
      RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));
    
      // create node
      RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));
    
      // create subscriber
      RCCHECK(rclc_subscription_init_default(
        &subscriber,
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Quaternion),
        "steppers"));
    
      // create executor
      RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
      RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
    
    
      for (int i = 0; i < numSteppers; i++) {
    
        pinMode(stepperPullPins[i],OUTPUT);
        pinMode(stepperDirPins[i],OUTPUT);
        pinMode(stepperEnablePins[i],OUTPUT);
        digitalWrite(stepperEnablePins[i],HIGH);
    
        stepper[i].setMaxSpeed(5000.0);
        stepper[i].setAcceleration(500.0);
        stepper[i].setSpeed(100.0);
    
        stepper[i].setMinPulseWidth(10);
      }
     
      delay(1000);
     
      Serial.begin(9600);
      Serial.println("Test:");
    
    }
    
    void loop() {
    
      RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
    
      for (int i = 0; i < numSteppers; i++) {
        stepper[i].moveTo(posObjectiu[i]);
        stepper[i].setSpeed(250.0);
        stepper[i].runSpeedToPosition();
      }
     
    }

Is there a way to call 

    RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));

that won't stop the steps from being send?

I learned I can use an ESP32 and utilize its second core, is that the only solution? Because changing from a teensy 4.1 to an ESP32 would cause me many more problems.

I also noticed that, if I create a publisher instead of a subscriber, the stepper is not affected in any way.

I would appreciate any help and thanks in advance.

lacie-life, 2025-03-08:
I meet the same problem. Have you solved it? @Marcal8

hippo5329, 2025-03-09:
You can switch to platformio/vscode. Combine the re-connection example and subscriber example. You can update the stepper position in subscriber or another control loop timer.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

lacie-life, 2025-03-09:
Thank you for your answer. I'm trying to use Platformio. But in my case, I use setSpeed function in AccelStepper. 
I also try to run the runSpeed() in another core, but when I try to set the priority of the task, with priority 0 and 1, the micro_ros node cannot connect to ESP32, and with priority 3, the connection restarts again and again. I don't know what happened. 
Is this a problem with micro_ros for ESP32?

The code I'm using is the following:

`#include <Arduino.h>
#include <micro_ros_arduino.h>
#include <AccelStepper.h>
#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/string.h>
#include <geometry_msgs/msg/twist.h>

#define WHEEL_RADIUS 0.150
#define DISTANCE_WHEEL 0.5
#define DIAMETER_WHEEL 0.3
#define STEP_PER_REVOLUTION 800
#define MOTOR_SPEED 1500

#define WHEEL_SEPARATION_X              0.6527
#define WHEEL_SEPARATION_Y              0.392
#define WHEEL_GEOMETRY                  (WHEEL_SEPARATION_X + WHEEL_SEPARATION_Y)

#define dir_pin_2 5
#define step_pin_2 18
#define dir_pin_1 22
#define step_pin_1 19
#define enable_1 23
#define enable_2 4
#define motorInterfaceType 1

int wheel_right_step = 0;
int wheel_left_step = 0;

AccelStepper motor_right_step(motorInterfaceType, step_pin_1, dir_pin_1);
AccelStepper motor_left_step(motorInterfaceType, step_pin_2, dir_pin_2);

rcl_node_t node;
rcl_publisher_t debug_publisher;
rcl_subscription_t cmd_vel_subscriber;
rclc_support_t support;
rcl_allocator_t allocator;
rclc_executor_t executor;

TaskHandle_t Core1Task; // Task handle for Core 1

std_msgs__msg__String debug_msg;
geometry_msgs__msg__Twist cmd_vel_msg;

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void driveStepper(void *param){
  motor_right_step.runSpeed();
  motor_left_step.runSpeed();
}

// Hàm callback xử lý dữ liệu cmd_vel
void commandVelocityCallback(const void *msg_in) {
    const geometry_msgs__msg__Twist *msg = (const geometry_msgs__msg__Twist *)msg_in;
    double linear_velocity = msg->linear.x;
    double angular_velocity = msg->angular.z;

    double wheel_right_kinematic = (linear_velocity + angular_velocity * (DISTANCE_WHEEL / 2)) / (DIAMETER_WHEEL / 2);
    double wheel_left_kinematic = (linear_velocity - angular_velocity * (DISTANCE_WHEEL / 2)) / (DIAMETER_WHEEL / 2);

    wheel_right_step = wheel_right_kinematic * STEP_PER_REVOLUTION / (2 * PI * WHEEL_RADIUS);
    wheel_left_step = wheel_left_kinematic * STEP_PER_REVOLUTION / (2 * PI * WHEEL_RADIUS);

    motor_right_step.setSpeed(wheel_right_step);
    motor_left_step.setSpeed(wheel_left_step);

    char debug_str[100];
    snprintf(debug_str, sizeof(debug_str), "Steps: %d, %d", wheel_right_step, wheel_left_step);
    debug_msg.data.data = debug_str;
    debug_msg.data.size = strlen(debug_str);
    debug_msg.data.capacity = sizeof(debug_str);

    rcl_publish(&debug_publisher, &debug_msg, NULL);
    
}
         
void setup() {

    xTaskCreatePinnedToCore(
      driveStepper,          // Function to implement the task
      "Core 1 Task",      // Name of the task
      10000,              // Stack size in words
      NULL,               // Task input parameter
      1,                  // Priority of the task
      &Core1Task,         // Task handle
      0                   // Run task on Core 1
    );

    set_microros_transports();
    delay(2000);

    allocator = rcl_get_default_allocator();
    rclc_support_init(&support, 0, NULL, &allocator);
    rclc_node_init_default(&node, "esp32_node", "", &support);
    rclc_publisher_init_default(&debug_publisher, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, String), "debug");
    rclc_subscription_init_default(&cmd_vel_subscriber, &node, ROSIDL_GET_MSG_TYPE_SUPPORT(geometry_msgs, msg, Twist), "cmd_vel");
    rclc_executor_init(&executor, &support.context, 2, &allocator);

    rclc_executor_add_subscription(&executor, &cmd_vel_subscriber, &cmd_vel_msg, &commandVelocityCallback, ON_NEW_DATA);

    pinMode(enable_1, OUTPUT);
    digitalWrite(enable_1, LOW);
    pinMode(enable_2, OUTPUT);
    digitalWrite(enable_2, LOW);
    
    motor_right_step.setMaxSpeed(MOTOR_SPEED);
    motor_right_step.setAcceleration(1000);
    motor_right_step.setSpeed(0);
    motor_left_step.setMaxSpeed(MOTOR_SPEED);
    motor_left_step.setAcceleration(1000);
    motor_left_step.setSpeed(0);
}

void loop() {
    // delay(10);
    // motor_right_step.runSpeed();
    // motor_left_step.runSpeed();
    RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10)));
}`

My config is:

- ROS Humble
- ESP32 Wroom

I would appreciate any help and thanks in advance.

hippo5329, 2025-03-10:
First esp32 core for micro-ros, and second core for stepper. You can not run ANY stepper code in first core. You may use global variables to pass position to the other core.