micro-ROS/micro_ros_arduino #1791: Want to rewrite the micro_ros_subscriber for example_interfaces/msg/String.

drinkBr, 2024-06-20:
Hi,
I want to rewrite the micro_ros_subscriber for example_interfaces/msg/String. 
However, when I tried to modify it myself, I encountered an error and couldn't proceed any further. I would like to know how to resolve this issue.
<The operating environment>
ArduinoIDE2.1.1
Arduino Due

program
<program><
```
#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <example_interfaces/msg/string.h>

rcl_subscription_t subscriber;
example_interfaces__msg__String msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void subscription_callback(const void * msgin)
{  
  const example_interfaces__msg__String * msg = (const example_interfaces__msg__String *)msgin;
  digitalWrite(LED_PIN, (msg->data = "0") ? LOW : HIGH);  
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create subscriber
  RCCHECK(rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(example_interfaces, msg, String),
    "key_hit_event"));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_subscription(&executor, &subscriber, &msg, &subscription_callback, ON_NEW_DATA));
}

void loop() {
  delay(100);
  RCCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}
```

Error message
```
rror: no match for 'operator=' (operand types are 'const rosidl_runtime_c__String' and 'const char [2]')
   digitalWrite(LED_PIN, (msg->data = "0") ? LOW : HIGH);  
                                    ^
```

Thank you.

hippo5329, 2024-06-20:
Is this a school assignment? You will need to solve it yourself.

Google "c++ string compare"

drinkBr, 2024-06-20:
Hi,hippo5329

Thank you for reply.
It's not school assignment ,but, I am a beginner with Arduino and ROS2, so this might be a simple question.
I thought that setting `msg->data = "\0"` would work, but what do you think?

My concern is not just about the syntax. In msg->data, msg is example_interfaces__msg__String, but data was rosidl_runtime_c__String. Since data was defined as 'String data' in example_interfaces__msg__String, I am puzzled as to why data is shown as belonging to a different class. I would appreciate your advice.

hippo5329, 2024-06-20:
digitalWrite(LED_PIN, (msg->data == "0") ? LOW : HIGH);  

The c++ string compare operator is "==" .

hippo5329, 2024-06-20:
Welcome to ROS2/mciro-ROS. There are a lot more puzzles. Enjoy. :)

As I suggested. It will save a lot of you if you install Ubuntu 22.04 PC, platformio/vscode, esp32 and follow my wiki. Workflow 1 and 2, should get you ready to explore the ROS world.

hippo5329, 2024-06-20:
Although it is possible to run ROS2 on Windows, it is more difficult for a beginner. You should get a spare laptop/desktop to install Ubuntu 22.04 and ROS2 humble.

hippo5329, 2024-06-20:
An old PC with 8GB memory will be fine. I used 10 years old PC myself. If you want to purchase new PC, be sure to have 32GB memory to build moveit2.

drinkBr, 2024-06-20:
Hey, hippo5329
Now I use ROS2 in Ubuntu22.04 on Raspberry Pi 4B.But, when I ran that program, used Arduino IDE on WindowsPC. 
Should I ran that program on Arduino IDE in Ubuntu?

hippo5329, 2024-06-20:
I would not suggest Arduino IDE. It is very limited on Pi. And it is awkward to build static micro-ros library on Arduino IDE. I would suggest install platformio/vscode on  your windows pc. It should work. I switched to Linux more than 30 years ago, and do not use windows since then. You should really setup an Ubuntu PC to learn and use ROS.

hippo5329, 2024-06-21:
Due has very limited memory. It uses very low memory meta. I would not recommend Due.

hippo5329, 2024-06-21:
I fired up a win11 vm and realized micro ros platformio won't work. 

You can install platformio cli on pi to build the micro-ros firmware.

platfromio.ini
```
[env:due]
platform = atmelsam
board = due
framework = arduino
board_microros_distro = humble
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
```
src/subcriber_twist.ino
```
#include <arduino.h>
#include <micro_ros_arduino.h>
...
void setup() {
  Serial.begin(115200);
  set_microros_serial_transports(Serial);
  // set_microros_transports();
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
...
```

drinkBr, 2024-06-21:
Do you mean downloading PlatformIO on Ubuntu running on a Raspberry Pi, instead of on Windows?

hippo5329, 2024-06-21:
Yes. Please follow my wiki to install platformio cli to pi. You may skip the linorobot2 and linorobot2_hardware installation.

I have created a repo for the example. You may clone and tryout.

  git clone https://github.com/hippo5329/micro-ros_subscriber_twist.git
  cd micro-ros_subscriber_twist
  pio run -t upload

hippo5329, 2024-06-21:
You may check the commit to understand the conversion.

https://github.com/hippo5329/micro-ros_subscriber_twist/commit/1018b3975b5abb7ba21a94b9e1819132da4aa1d2

hippo5329, 2024-06-21:
However, it will save some efforts in the future if you install the two lino projects.

hippo5329, 2024-06-21:
Due has very limited memory. You should get some esp32. It has 5 times the RAM  and 8 times the Flash ROM of Due. The CPU speed is much faster on esp32, too. The WifI support of esp32 is another plus.

drinkBr, 2024-06-21:
When I executed 'pio run -t upload', the following error occurred.

The actions I took were:
`curl -fsSL -o get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
python3 get-platformio.py
curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules
sudo service udev restart
sudo usermod -a -G dialout $USER
sudo usermod -a -G plugdev $USER
git clone https://github.com/hippo5329/micro-ros_subscriber_twist.git
cd micro-ros_subscriber_twist
pio run -t upload`

Is it okay to proceed with this?

hippo5329, 2024-06-21:
Please complete the software installation in my wiki. Do not skip any steps.

The steps to install platformio as in my wiki.

sudo apt remove brltty -y
sudo apt install python3-venv build-essential cmake git curl -y

curl -fsSL -o get-platformio.py https://raw.githubusercontent.com/platformio/platformio-core-installer/master/get-platformio.py
python3 get-platformio.py

echo "PATH=\"\$PATH:\$HOME/.platformio/penv/bin\"" >> ~/.bashrc
source ~/.bashrc

curl -fsSL https://raw.githubusercontent.com/platformio/platformio-core/develop/platformio/assets/system/99-platformio-udev.rules | sudo tee /etc/udev/rules.d/99-platformio-udev.rules
sudo service udev restart
sudo usermod -a -G dialout $USER
sudo usermod -a -G plugdev $USER

hippo5329, 2024-06-21:
You may be confused in the beginning. But in the future, you will need to understand what and why every steps in my wiki.

hippo5329, 2024-06-21:
Updated. Please check README.

https://github.com/hippo5329/micro-ros_subscriber_twist

hippo5329, 2024-06-22:
My previous replies on string msg are wrong.

String is complex datatype and needs special memory allocation. Please check this,
https://github.com/hippo5329/micro-ROS-demos-platformio/blob/rolling/rclc/string_subscriber/src/main.c

digitalWrite(LED_PIN, strcmp(msg->data.data,  "0") ? HIGH : LOW);

There are more examples in micro_ros_demos. You may clone my repo and try out.

https://github.com/hippo5329/micro-ROS-demos-platformio

I get a Due and give it 20 mins try-out with both int32_publisher and string_publisher. I am not able to git it connect to mico-ROS after upload. I give up because I am not interested in Due.

drinkBr, 2024-06-22:
To install platformio,I followed all the steps, but the following error occurred.

```
Command 'pio' not found, but can be installed with:
sudo apt install platformio
```
"After running `sudo apt install platformio`, it seemed to work. When I ran `pio run -e due -t upload`, the following result occurred."

hippo5329, 2024-06-22:
Did you create the sd image with rpi-imager? Which OS version had you chosen? You should use ubuntu 22.04 ltd desktop or server. Do not use core version.

You may use ubuntu lts 24.04 and ROS2 jazzy. It will get better support in the future.

hippo5329, 2024-06-22:
Please check your platformio installation with

  ls ~/.platformio/penv/bin

If you can find pio file here, then it is problem in the PATH. Try

  echo $PATH

You should find the above path included.

hippo5329, 2024-06-22:
I checked again on a rpi3b ubuntu 22.04 lts server 64 bits. There is no issue in pio installation.

https://github.com/hippo5329/micro-ROS-demos-platformio#install-platformio

hippo5329, 2024-06-25:
I forced pushed update. Due should work with most example packages now. Please remove old dir and clone again. Some won't work due to Due limitation.

hippo5329, 2024-06-25:
The arduino sam port has not been updated for 4 years. Issues remains unresolved. Arduino Due should be avoided.

drinkBr, 2024-06-25:
I'm sorry for the late reply.
> Which OS version did you choose?
I chose Ubuntu Desktop 22.04 LTS(64bit). So, I used ROS2 humble.

> Please check your platformio installation with
> ls ~/.platformio/penv/bin
> If you can find pio file here, then it is problem in the PATH. Try
> echo $PATH
> You should find the above path included.
The results of the execution are as follows.

What should I do? Should I start over from the beginning with the PlatformIO procedures?
``
Also, I got ESP32. It says ESP-WROOM-32, but is it compatible with your wiki?

drinkBr, 2024-06-25:
Also, I have one more question.
In micro_ros_setup,

hippo5329, 2024-06-25:
The PATH setup was wrong. Please edit your .bashrc, remove all lines containing "~/.platformio/penv/bin" . Then add this line to the end,

    PATH="$PATH:$HOME/.platformio/penv/bin"

Then open a new terminal, and check with, "which pio". It should be like "/home/ubuntu/.platformio/penv/bin/pio" , here ubuntu is your user name. The pio command should be available now.

The ESP-WROOM-32 works well.

If you follow my wiki, no need to worry about micro ros setup. It is included in the linorobot2 installation. The micro-ros agent will be built and installed. You can follow the steps up to check topics with a WROOM esp32 module and wfi transport. This is a setup to help you learn micro-ros. You do not need to use the firmware in the future. Wifi transport is very helpful to run the various micro-ros demos.

https://github.com/hippo5329/linorobot2_hardware/wiki#install-linorobot2
https://github.com/hippo5329/linorobot2_hardware/wiki#install-firmware-source
https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-serial-transport-on-devttyusb0
https://github.com/hippo5329/linorobot2_hardware/wiki#check-topics
https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-wifi-transport

hippo5329, 2024-06-25:
For your information,

https://github.com/hippo5329/linorobot2_hardware/wiki#build-micro-ros-agent-alone---optional

You do not need this. The linorobot2 installation will build micro-ROS agent.

drinkBr, 2024-06-27:
> Please edit your .bashrc, remove all lines containing "~/.platformio/penv/bin" . Then add this line to the end,
> 
> PATH="$PATH:$HOME/.platformio/penv/bin"
> Then open a new terminal, and check with, "which pio". It should be like "/home/ubuntu/.platformio/penv/bin/pio" , here ubuntu is your user name.

After executing this, the results of executing steps `pio run -e due -t upload` and `pio run -e esp32 -t upload` are as follows.

Is PlatformIO functioning correctly with this? Should I start over from the beginning?

hippo5329, 2024-06-27:
It is because you installed platformio with apt, which is /usr/bin/pio. While we install platformio with python penv. You may try "sudo apt remove platformio". Or you may start over with a fresh ubuntu 22.04. You will need to execute those commands I listed in my wiki. No need to follow the links to upstream sites. I should remove those link to external sites, such as platformio. It causes confusion.

hippo5329, 2024-06-27:
I have removed those confusing external links. You may start here on a fresh ubuntu 22.04 rpi4,

https://github.com/hippo5329/linorobot2_hardware/wiki#install-the-software

drinkBr, 2024-06-27:
After executing `sudo apt remove platformio`, I executed `pio run -e esp32 -t upload`, and the results are as follows.

Is this successful?

drinkBr, 2024-06-27:
How should I write 'cd firmware' when moving from the home directory as described in 'esp32 with micro-ROS serial transport on /dev/ttyUSB0'?

https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-serial-transport-on-devttyusb0

hippo5329, 2024-06-27:
Good. The esp32 built and uploaded successfully. You may proceed to test micro ROS connection.

cd linorobot2_hardware/firmware
pio run -e esp32 -t upload
ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600

drinkBr, 2024-06-28:
After executing step 1, the following message was displayed

Is this successful?
I have more question about commands in Terminal.
I think `pio run -e esp32 -t upload` writes the program to the ESP32.` ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 --baudrate 921600` connects the ESP32 to ROS2.
This is what I think; is it correct?

Also, I believe the program written in `pio run -e esp32 -t upload` is firmware.cpp located in ~/linorobot2_hardware/firmware/src. Is it possible to replace this with another program, such as micro_ros_subscriber_twist.ino? The content of firmware.cpp is very complex and difficult for me to understand, so I would like to know what each phase is doing.

hippo5329, 2024-06-28:
> After executing step 1, the following message was displayed
> Is this successful? I have more question about commands in Terminal. I think `pio run -e esp32 -t upload` writes the program to the [...]

It is successful.
Correct.

> Also, I believe the program written in `pio run -e esp32 -t upload` is firmware.cpp located in ~/linorobot2_hardware/firmware/src. Is it possible to replace this with another program, such as [...]

You should proceed to check to topics. Google the massages types. Make sure you understand the commands and messages.
https://github.com/hippo5329/linorobot2_hardware/wiki#check-topics

Then proceed to wifi transport. Connect via wifi transport and check topic again.
https://github.com/hippo5329/linorobot2_hardware/wiki#esp32-with-micro-ros-wifi-transport

After checked topics and wifi transport, you may leave the linorobot2_hardware. And proceed to other examples.

1. the ported subscribe twist example from micro_ros_arduino.
https://github.com/hippo5329/micro-ros_subscriber_twist

2. the ported micro_ros_demos.
https://github.com/hippo5329/micro-ROS-demos-platformio

> One more question please. .cpp files can be edited with a text editor, but .ino files can currently only be edited with the Arduino IDE. Is this okay?

A .ino file can be edited with any text editor. In file manager, select file, right mouse click, "open with" Text Editor.
Or you may install gedit, an easy-to-use and general-purpose text editor.

    sudo apt install -y gedit

hippo5329, 2024-06-28:
OR you may install vscode on rpi4. Or install vscode on windows, and use remote development extension ssh into pi.

hippo5329, 2024-06-29:
https://marketplace.visualstudio.com/items?itemName=ms-vscode-remote.remote-ssh
https://carleton.ca/scs/2023/vscode-remote-access-and-code-editing/

drinkBr, 2024-06-29:
> You should proceed to check to topics. Google the massages types. Make sure you understand the commands and messages.
> https://github.com/hippo5329/linorobot2_hardware/wiki#check-topics

I checked topics. They're correct.

> 1. the ported subscribe twist example from micro_ros_arduino. https://github.com/hippo5329/micro-ros_subscriber_twist

This did not work, and the following error was found.

I changed the baud rate of micro-ros_subscriber_twist.ino to 921600, but it did not resolve the issue. This is probably unrelated to the issue.
I think the cause of this issue is as follows.

1. The code to be uploaded is not in /linorobot2_hardware.
2. The file format is .ino.
3. The micro-ros_subscriber_twist I am using was cloned from git last week.

I would like to hear your thoughts.

hippo5329, 2024-06-29:
> > You should proceed to check to topics. Google the massages types. Make sure you understand the commands and messages.
> > https://github.com/hippo5329/linorobot2_hardware/wiki#check-topics
> 
> I checked topics. They're correct.

Good.

> 
> > 1. the ported subscribe twist example from micro_ros_arduino. https://github.com/hippo5329/micro-ros_subscriber_twist
> 
> This did not work, and the following error was found.
> 
> ```
> Linking .pio/build/esp32/firmware.elf
> ~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0x8): undefined reference to `setup()'
> ~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libFrameworkArduino.a(main.cpp.o):(.literal._Z8loopTaskPv+0xc): undefined reference to `loop()'
> ~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32/libFrameworkArduino.a(main.cpp.o): in function `loopTask(void*)':
> ~/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42: undefined reference to `setup()'
> ~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /home/agri/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:48: undefined reference to `loop()'
> collect2: error: ld returned 1 exit status
> *** [.pio/build/esp32/firmware.elf] Error 1
> ===================================== [FAILED] Took 31.65 seconds =====================================
> 
> Environment    Status    Duration
> -------------  --------  ------------
> esp32          FAILED    00:00:31.650
> ```
> 
> It seems like it doesn't recognize setup() and loop(). I changed the baud rate of micro-ros_subscriber_twist.ino to 921600, but it did not resolve the issue. This is probably unrelated to the issue. I think the cause of this issue is as follows.
> 
>     1. The code to be uploaded is not in /linorobot2_hardware.

It is correct. It is not related to linorobot2_hardware. It should be outside linorobot2_hardware.

> 
>     2. The file format is .ino.

It does not matter the file extension is .ino or .cpp .

> 
>     3. The micro-ros_subscriber_twist I am using was cloned from git last week.

Please remove the old clone directory and clone again.

> 
> 
> I would like to hear your thoughts.
```
Building micro-ROS library
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- micro_ros_platformio @ 0.0.1+sha.25ab659
Building in release mode
Compiling .pio/build/esp32/src/micro-ros_subscriber_twist.ino.cpp.o
Building .pio/build/esp32/bootloader.bin
Generating partitions .pio/build/esp32/partitions.bin
Compiling .pio/build/esp32/libe28/micro_ros_platformio/platform_code/arduino/clock_gettime.cpp.o
esptool.py v4.5.1
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Compiling .pio/build/esp32/libe28/micro_ros_platformio/platform_code/arduino/serial/micro_ros_transport.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/FirmwareMSC.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/HWCDC.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/HardwareSerial.cpp.o
Archiving .pio/build/esp32/libe28/libmicro_ros_platformio.a
Indexing .pio/build/esp32/libe28/libmicro_ros_platformio.a
[... 36 lines cut ...]
Compiling .pio/build/esp32/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/esp32/FrameworkArduino/main.cpp.o
Compiling .pio/build/esp32/FrameworkArduino/stdlib_noniso.c.o
Compiling .pio/build/esp32/FrameworkArduino/wiring_pulse.c.o
Compiling .pio/build/esp32/FrameworkArduino/wiring_shift.c.o
Archiving .pio/build/esp32/libFrameworkArduino.a
Indexing .pio/build/esp32/libFrameworkArduino.a
Linking .pio/build/esp32/firmware.elf
Retrieving maximum program size .pio/build/esp32/firmware.elf
Checking size .pio/build/esp32/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.4% (used 40772 bytes from 327680 bytes)
Flash: [===       ]  25.6% (used 335649 bytes from 1310720 bytes)
Building .pio/build/esp32/firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
======================== [SUCCESS] Took 368.52 seconds ========================

Environment    Status    Duration
-------------  --------  ------------
esp32          SUCCESS   00:06:08.515
========================= 1 succeeded in 00:06:08.515 =========================
```

hippo5329, 2024-06-29:
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

drinkBr, 2024-06-30:
> ```
> Linking .pio/build/esp32/firmware.elf
> ~/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: [...]

When editing the .ino file using the Arduino IDE, a new folder is created in the location where the file was, and the edited file is saved in that folder. Is this okay? I believe this is the reason why setup() and loop() were not recognized.

drinkBr, 2024-06-30:
As a test, I changed the LED_PIN in micro-ros_publisher to 4 and uploaded the program, but the attached LED did not light up. Why is this? The positive side of the LED is connected to "D4" and the negative side to "GND". The resistor is 1kΩ. I realized while writing that the resistor might be too large.

Also, should the pin numbers specified in the program be the same as the pin numbers written on the ESP32 itself? I researched this but couldn't find a clear answer.

hippo5329, 2024-06-30:
> When editing the .ino file using the Arduino IDE, a new folder is created in the location where the file was, and the edited file is saved in that folder. Is this okay? I believe this is the reason why setup() and loop() were not recognized.

So the arduino ide should not be used to edit this project. Arduino will interfere with our build process.  Please install gedit and use gedit to edit your project. Remove the old directory and clone again.

hippo5329, 2024-06-30:
> As a test, I changed the LED_PIN in micro-ros_publisher to 4 and uploaded the program, but the attached LED did not light up. Why is this? The positive side of the LED is connected to "D4" and the [...]

I have changed the LED_PIN in my port.

```
#ifdef LED_BUILTIN
#define LED_PIN LED_BUILTIN
#else
#define LED_PIN 13
#endif
```

I used LED_BUILTIN, which is 2 in most esp32 modules, including wroom.  Wroom has two LEDs on board. The red led is power. The blue led is 2. You should see blue led lights up.

hippo5329, 2024-06-30:
A Blink LED example in platformio.

https://github.com/hippo5329/Blink-platformio

hippo5329, 2024-06-30:
I think our conversations has gone far away from the scope of this repository. Please open a new issue in my repo,
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/issues

drinkBr, 2024-06-30:
I have created a new issue. Thank you.