micro-ROS/micro_ros_arduino #1804: Having trouble when running micro-ros_publisher_wifi example

winster-bai, 2024-07-16:
## Issue template

- Hardware description: ESP32
- Installation type: micro_ros_setup
- Version or commit hash: Humble

#### Steps to reproduce the issue
First of all, I installed the micro_ros library file according to the tutorial. 
The esp32 is connected to the computer via USB. It runs normally on the example code "micro-ros_publisher.ino" and the host can receive the signal sent by the esp32. 
But when I uploaded the micro-ros_publisher_wifi.ino code (the ssid and pwd in the code have been modified), I ran 
"ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888" 
on the host, but this time i did not receive any data.
#### Expected behavior

I have ensured that the ssid and pwd are correct, the IP address is consistent with the host IP, and there is no new topic in the ros2 topic list. Photos are as followed:
[image]

Can you help me? Thanks in advance

hippo5329, 2024-07-16:
The esp32 is not connected to the agent. You may follow this wiki to use platformio to try out the examples.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

hippo5329, 2024-07-16:
The set_microros_wifi_transports() will wait until wifi connected. If LED does not light, it might be esp32 wifi connection issue.

winster-bai, 2024-07-17:
> The set_microros_wifi_transports() will wait until wifi connected. If LED does not light, it might be esp32 wifi connection issue.set_microros_wifi_transports() 将等待 wifi 连接。如果 LED 不亮，则可能是 esp32 wifi 连接问题。

Thanks for reply me, i used command "ros2 run micro_ros_setup create_agent_ws.sh" to creat the agent. The LED light of esp32 has been lit after connecting to wifi, but I still cannot connect to the device through udp on the host.

hippo5329, 2024-07-17:
You may try "ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6" to see more debug messages.

Or, try my wiki.

winster-bai, 2024-07-17:
> You may try "ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6" to see more debug messages.
> 
> Or, try my wiki.

Hi, thank you for the reply;
There is no more output using "ros2 run micro_ros_agent micro_ros_agent udp4 --port 8888 -v6". 
Yes I tried the wiki steps earlier, but unfortunately, it seemed that some steps conflicted with the humble environment I used before, which resulted in no success in the end. I will try to get a new linux environment and try your wiki again, or abandon arduino and use esp-idf. 
Anyway, thanks for helping me.

closed 2024-07-17 by winster-bai