micro-ROS/micro_ros_arduino #1810: esp32 publishing rate limited at 2 Hz

amangan, 2024-07-25:
I am not able to increase the frequency more than 2 Hz (seems like a similar issue to [this one](https://github.com/micro-ROS/micro_ros_arduino/issues/56)). It publishes two quickly data points then pauses

Description: 
- esp 32 Ubuntu
- OS - Ubuntu 22.04.4 LTS
- micro_ros_arduino humble branch (pulled today)
- - Uploading via arduino IDE

#### Steps to reproduce the issue
Install micro_ros_arduino humble 
try to decrease timer_timeout in publisher example (it is able to increase & is printing the expected values when connecting to an encoder)
echo output
when running `ros2 topic hz /micro_ros_arduino_node_publisher` getting the output below

Even when trying to use rclc_client_init_best_effort, I am seeing the same output. Same thing when getting rid of the delay in the loop()

[image]

Any ideas on things to try or know of common causes?

hippo5329, 2024-07-25:
You may try remove the delay(100) in the loop.

void loop() {
// delay(100);
RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

amangan, 2024-07-25:
Yeah I mention above that the same output occurs when getting rid of the delay in the loop() and when using clc_client_init_best_effort

hippo5329, 2024-07-26:
Please follow my wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki.

Remove the delay. I can run to 200Hz without issue. The max rate is about 250Hz with my i7 laptop.

amangan, 2024-07-29:
@hippo5329 just followed your wiki (but for humble) and everything seems to be working - thank you!

closed 2024-07-29 by amangan