micro-ROS/micro_ros_arduino #1811: After isolating micro ROS topics, the data is not being published

Muthukumar4796, 2024-07-26:
**Device - Arduino Due**

After isolate the micro ros topic using,

/ Initialize micro-ROS allocator
rcl_allocator_t allocator = rcl_get_default_allocator();

// Initialize and modify options (Set DOMAIN ID to 10)
rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
rcl_init_options_init(&init_options, allocator);
rcl_init_options_set_domain_id(&init_options, 10);

// Initialize rclc support object with custom options
rclc_support_t support;
rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator);

**The micro ROS topic is not publishing. How can I resolve this issue? I have shared my code and attached screenshot for reference**

#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  // Initialize micro-ROS allocator
  rcl_allocator_t allocator = rcl_get_default_allocator();
  
  // Initialize and modify options (Set DOMAIN ID to 7)
  rcl_init_options_t init_options = rcl_get_zero_initialized_init_options();
  rcl_init_options_init(&init_options, allocator);
  rcl_init_options_set_domain_id(&init_options, 7);
  
  // Initialize rclc support object with custom options
  rclc_support_t support;
  rclc_support_init_with_options(&support, 0, NULL, &init_options, &allocator);

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1000;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 0;
}

void loop() {
  delay(100);
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
}

[image]

hippo5329, 2024-07-26:
What do you mean by "isolate"? Do you try topic echo?

Muthukumar4796, 2024-07-26:
Isolate means trying to avoid floating topics in the same network. Can you check the attached screenshot. I tried to echo the topic

hippo5329, 2024-07-26:
Have you tried Jazzy or rolling? I don't know if isolation is implemented in micro-ROS.

hippo5329, 2024-07-26:
You may use namespace, prefix to topic. Or set ROD_DOMAIN to an agent.

hippo5329, 2024-07-26:
micro-ROS-demos rclc configuration_ecamples

hippo5329, 2024-07-28:
I  tried the configuration. The micro-ros agent is not connected. The same as your result. Will need dig into rclc.

hippo5329, 2024-07-29:
Did you set the domain_id with "export ROS_DOMAIN_ID=7" before echo?

I repeated the test with domain_id as 10. It works.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/commit/75bec5c265880a2a1409fb40407c372e57f4df69

Muthukumar4796, 2024-07-29:
Its Working. Thanks