micro-ROS/micro_ros_arduino #1812: micro-ros agent only detects the USB after unplugging and plugging back the USB

srujan-xairo, 2024-07-28:
## Issue template

- Hardware description: 
Raspberry Pi 4 Model B

- RTOS: 
Using micro-ROS with Raspberry Pi Pico. 

- Installation type: 
Followed the installation steps from **Teensy with Arduino** section from the Tutorials of micro-ros website.

- Version or commit hash: 
Using humble

#### Actual behavior
when I run the command 
` ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0`

I get the below output

```
[1722173514.344108] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1722173514.345154] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
```
and does nothing. After I unplug the USB and Plug it back I get the below output and all my nodes are active. Which is the expected behavior, but why do I need to unplug and plug the USB every time? How can I solve this?

```
[1722173654.738771] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1722173656.288891] info     | Root.cpp           | create_client            | create                 | client_key: 0x0DA77874, session_id: 0x81
[1722173656.289134] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x0DA77874, address: 0
[1722173656.328959] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x0DA77874, participant_id: 0x000(1)
[1722173656.331903] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0DA77874, topic_id: 0x000(2), participant_id: 0x000(1)
[1722173656.333897] info     | ProxyClient.cpp    | create_subscriber        | subscriber created     | client_key: 0x0DA77874, subscriber_id: 0x000(4), participant_id: 0x000(1)
[1722173656.336467] info     | ProxyClient.cpp    | create_datareader        | datareader created     | client_key: 0x0DA77874, datareader_id: 0x000(6), subscriber_id: 0x000(4)
[1722173656.339489] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0DA77874, topic_id: 0x001(2), participant_id: 0x000(1)
[1722173656.341205] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0DA77874, publisher_id: 0x000(3), participant_id: 0x000(1)
[1722173656.344010] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0DA77874, datawriter_id: 0x000(5), publisher_id: 0x000(3)
[1722173656.346707] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0DA77874, topic_id: 0x002(2), participant_id: 0x000(1)
[1722173656.348612] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0DA77874, publisher_id: 0x001(3), participant_id: 0x000(1)
[1722173656.351384] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0DA77874, datawriter_id: 0x001(5), publisher_id: 0x001(3)
[1722173656.353954] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0DA77874, topic_id: 0x003(2), participant_id: 0x000(1)
[1722173656.355947] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0DA77874, publisher_id: 0x002(3), participant_id: 0x000(1)
[1722173656.358593] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0DA77874, datawriter_id: 0x002(5), publisher_id: 0x002(3)
[1722173656.361138] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x0DA77874, topic_id: 0x004(2), participant_id: 0x000(1)
[1722173656.363170] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x0DA77874, publisher_id: 0x003(3), participant_id: 0x000(1)
[1722173656.365905] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x0DA77874, datawriter_id: 0x003(5), publisher_id: 0x003(3)
```

hippo5329, 2024-07-28:
You will need to add re-connection support.

https://github.com/micro-ROS/micro_ros_arduino/blob/jazzy/examples/micro-ros_reconnection_example/micro-ros_reconnection_example.ino

Use of Humble is not recommended for new designs. You should move to current LTS stable release Jazzy and Ubuntu 24.04.

hippo5329, 2024-07-28:
You may check my wiki. 
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

hippo5329, 2024-07-28:
You should check if /dev/ttyACM0 exists. "ls /dev/ttyA*" .

The arduino examplea without re-connection will timeout and enter error_loop(), if it failed to connect the agent.

srujan-xairo, 2024-08-15:
This solved my issue. Thanks.

closed 2024-08-15 by srujan-xairo