micro-ROS/micro_ros_arduino #1813: Problems running ping-pong application on Arduino DUE and Serial communication

Latek, 2024-07-30:
## Issue template

- Hardware description: Macbook Pro + Arduino DUE
- Software description: Ubuntu Linux 22.04.1 installed on Virtualbox, installed ROS2 Humble according to the description: https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html, Installed Micro ROS according to the description: https://micro.ros.org/docs/tutorials/core/first_application_linux/
- RTOS: Arduino IDE
- Installation type: micro_ros_setup
- Version or commit hash: 

#### Steps to reproduce the issue
1. Compiled and uploaded the following code using the Arduino IDE + micro_ro_arduino plugin using the patch for Arduino DUE:
micro_ros_arduino_ping_pong.ino:
```
#include <Arduino.h>
#include <micro_ros_arduino.h>

#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/header.h>

#include <stdio.h>
#include <time.h>
#include "clock_gettime.h"

#define STRING_BUFFER_LEN 100
#define LED_PIN 13

#define RCCHECK(fn)                                                                      \
    {                                                                                    \
        rcl_ret_t temp_rc = fn;                                                          \
        if ((temp_rc != RCL_RET_OK))                                                     \
        {                                                                                \
            printf("Failed status on line %d: %d. Aborting.\n", __LINE__, (int)temp_rc); \
            error_loop();                                                                \
        }                                                                                \
    }
#define RCSOFTCHECK(fn)                                                                    \
    {                                                                                      \
        rcl_ret_t temp_rc = fn;                                                            \
        if ((temp_rc != RCL_RET_OK))                                                       \
        {                                                                                  \
            printf("Failed status on line %d: %d. Continuing.\n", __LINE__, (int)temp_rc); \
        }                                                                                  \
    }

void error_loop()
{
    while (1)
    {
        digitalWrite(LED_PIN, !digitalRead(LED_PIN));
        delay(100);
    }
}

rclc_executor_t executor;
rcl_publisher_t ping_publisher;
rcl_publisher_t pong_publisher;
rcl_subscription_t ping_subscriber;
rcl_subscription_t pong_subscriber;

std_msgs__msg__Header incoming_ping;
std_msgs__msg__Header outcoming_ping;
std_msgs__msg__Header incoming_pong;

int device_id;
int seq_no;
int pong_count;

[... 192 lines cut ...]
[ERROR] [1722378449.432044514] [rclc]: [rclc_init] Error in rcl_init: error not set, at ./ erc/rcl/init.c:218

Failed status on line 82: 1 Aborting.
[ros2run]: Process exited with failure 1
```

#### Case 2:
1. When running
```
export RMW_IMPLEMENTATION=
ros2 run micro_ros_demos ping_pong
```
(RMW_IMPLEMENTATION set to an empty value)

#### Expected behavior 2:
See the ping-pong responding back and forth as in the example https://micro.ros.org/docs/tutorials/core/first_application_linux/, Multiple Ping Pong nodes

#### Actual behavior 2:
```
[INFO] [1722378409.760130758] [] Created a timer with period 2000mc.

Ping send seq 846930886_1804289383
realloc(): invalid pointer
[ros2run]: Aborted
```

hippo5329, 2024-07-31:
Since you are using micro-ros serial transport, you cannot use serial printf at the same time. You should comment out the printf.

We are using the default fast dds. There is no need to set the RMW_IMPLEMENTATION. And rmw_microxrcedds should not be used with arduino.

Have you tried the int32 publisher? You should start with the simple one.

hippo5329, 2024-07-31:
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wik

Latek, 2024-08-02:
Thank you for your response.

> Since you are using micro-ros serial transport, you cannot use serial printf at the same time. You should comment out the printf.

You're 100% right! My mistake!
> 
> We are using the default fast dds. There is no need to set the RMW_IMPLEMENTATION. And rmw_microxrcedds should not be used with arduino.
> 
> Have you tried the int32 publisher? You should start with the simple one.

Int32 publisher works as expected. When I removed printf calls my ping-pong example still doesn't work and I got the same error: "realloc(): invalid pointer [ros2run]: Aborted". Sometimes the ping is sent before the error, sometimes not. 

I've also tried the micro-ros_addtwoints_service example where the agent seems to connect to due, but the ros2 service caller waits endlessly for the service to be available.

Is the Serial transport method not designed to be used in all cases/examples? Do you think the problem is with using Arduino DUE?

I have built the robot based on the Ardumower (https://wiki.ardumower.de/) for which I'd like to rewrite the program using ROS2 + micro-ROS. The PCB 1.4 is compatible with Arduino DUE, Mega, or Adafruit Grand Central M4 Express.

hippo5329, 2024-08-02:
arduino due $50 single core 84Mhz 512K flash 96K sram, Arm will stop mbed support soon.
esp32 $5 dual core 240MHnz 4M flash 512k sram wifi on chip, well supported freertos/arduino/micro-ros

due has very small flash and sram. it has to use the low memory colcon meta. due has very limited capability for micro-ros.
You should really switch to esp32.

hippo5329, 2024-08-02:
https://github.com/hippo5329/linorobot2_hardware/wiki