micro-ROS/micro_ros_arduino #1814: is arduino giga r1 supported for micro-ros?

adijams01, 2024-08-01:
## Issue template

- Hardware description: arduino giga r1
- RTOS: 
- Installation type: 
- Version or commit hash: foxy/humble

#### Steps to reproduce the issue
will micro-ROS work with this board https://store-usa.arduino.cc/products/giga-r1-wifi?selectedStore=us?

#### Expected behavior

#### Actual behavior

#### Additional information

hippo5329, 2024-08-01:
If you are new to micro-ros, I would suggest starting with esp32 which is much cheaper and better supported.

hippo5329, 2024-08-02:
arduino due $50 single core 84Mhz 512K flash 96K sram, arm give up mbed support
esp32 $5 dual core 240MHnz 4M flash 512k sram wifi on chip, well supported freertos/arduino/micro-ros

due has very small flash and sram. it will need the low memory colcon meta. due has very limited capability for micro-ros.
You should really switch to esp32.

adijams01, 2024-08-05:
Previously I was working with rosserial with Arduino mega in ROS noetic, I was thinking to upgrade to humble with Arduino giga R1.
My project requires lot of pins and dual core so I wanted to know if it is possible to work with Arduino giga R1?

gbr1, 2024-11-14:
Last PR supports Arduino Giga R1

adijams01, 2024-11-14:
Amazing @gbr1 
Thank you for the update

closed 2025-07-15 by 4ntn