micro-ROS/micro_ros_arduino #1817: esp32 can only publish up to 2hz. slow performance

TheHassanShahzad, 2024-08-12:
## Issue template

- Hardware description: esp32 wroom32 nodeMCU dev board
- Installation type: downloaded zip library and included it in arduino IDE
- Version or commit hash: humble running microros arduino 2.0.7

#### Steps to reproduce the issue
i modified the example micro ros publisher which should publish an integer regulary. I modified the timer period to 20ms (anything under 600 doesnt seem to work as expected) and got rid of the 100ms delay in the main loop. 

After uploading the code, i ran the microROS Agent which i cloned into my src and built then ran ' ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0'

#### Expected behavior
ros2 topic echo /topic_name should display the integers being printed at around 50Hz 
#### Actual behavior
https://asciinema.org/a/D1Z1pUeNKVQrnGEfAYVl7WP9p

integers published up to 2hz and are laggy. get published together instead of in evenly spaced intervals
#### Additional information
Ive used microROS on esp32 with microROS agent and have never had any issue. Only until i started adding custom message types did these issues start

Im using rmw_fastrtps_cpp but even switching to rmw_cyclonedds_cpp did not solve the issue
Plugging the esp32 into a raspberry pi and running the agent also did not fix the issue so i think the issue lies with the library and not the agent

pehaps a fault in a new update

hippo5329, 2024-08-13:
You will need to change the baudrate  to 921600. The maximum topic rate of int32 publisher is around 250Hz on esp32. I use 50Hz for control loop. You should also use low latency kernel on host. Please follow my wiki.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

TheHassanShahzad, 2024-08-13:
I already followed your tutorial prior to my issue. I changed the kernel to be low latency and changed the baud rate but still no solution.  I found that i can send data to the esp32 at very high frequencies with no problems but it is slow at acting like a publisher and publishing data

Could that be of any help in diagnosing the issue

hippo5329, 2024-08-13:
Have you tried the simple int32 publisher? Does it work at higher rate?

TheHassanShahzad, 2024-08-14:
That's what I'm trying. The int32 publisher at default works perfectly fine but when I change the timer callback to be faster say 10Hz then it doesn't work

hippo5329, 2024-08-14:
Then this is the problem. I tested again on my old i7-4770k ubuntu 24.04 desktop. The esp32 int32 publisher can work at 500Hz.

What is your hardware/software environment?

TheHassanShahzad, 2024-08-14:
ok it might be worth trying it on a fresh install of ubuntu 22.04 for testing and development. I tried testing it on my raspberry pi with fairly few programs on it and still no change

On the laptop im running intel  i7 1360p with ubuntu 22.04.4. Ive tried microROS before with this setup and it worked fine. I think i installed/deleted something which is now causing some problems and i need to diagnose what that is

hippo5329, 2024-08-14:
You should switch to ubuntu 24.04 and ros2 jazzy. They are the latest LTS stable. There are a lot of improvements.

sumomomomomomomomo, 2024-10-08:
I encountered a similar issue with the ESP32-WROOM-32 and ESP32S3. 
In my case, the problem was resolved by downgrading the version of the ESP32 board package in the Arduino IDE from 3.0.5 to 2.0.18. 
My environment is Ubuntu 22.04 with ROS 2 Humble.

xopxe, 2024-10-18:
I'm seeing the same problem with a ESP32 PICO-D4 board. With eps32 libraries v2.0.18 works fine, with 3.0.5 timers misbehave badly.

hippo5329, 2024-10-19:
The micro-ROS-arduino esp32 library is compiled with v2.0.x. It might not work with arduino-esp32 v3.

batmanbetav, 2025-02-27:
> I encountered a similar issue with the ESP32-WROOM-32 and ESP32S3. In my case, the problem was resolved by downgrading the version of the ESP32 board package in the Arduino IDE from 3.0.5 to 2.0.18. My environment is Ubuntu 22.04 with ROS 2 Humble.

I did the same but I'm now stuck with 10hz, the speed doesn't seem to go beyond that