micro-ROS/micro_ros_arduino #1818: How to Loading Firmware to esp32 if my project has 2 file .c ?

flux0-0, 2024-08-17:
My project has 3 file, i do " ros2 run micro_ros_setup configure_firmware.sh myproject --transport serial "  - Successfully 
" ros2 run micro_ros_setup flash_firmware.sh " - Successfully  , "ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyUSB0 "  - Successfully . But only node, topic from the app.c  created, and led_control.c is not. How to fix it ? 
P/s: My project save in folder FreertosApp/Firmware

[image]

hippo5329, 2024-08-17:
micro-ros-platformio is a good alternaive to micro-ros-setup. All files in project src directory will be compiled. You may follow my wiki. https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

flux0-0, 2024-08-18:
> micro-ros-platformio is a good alternaive to micro-ros-setup. All files in project src directory will be compiled. You may follow my wiki. https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

 i plan to use esp-idf with platformio and microros, do you have this tutorial ? And What option do you suggest  for a robot car project using cameras and lidar? Esp-idf or adruino ?

hippo5329, 2024-08-18:
Arduino has more drivers support than esp-idf. I would suggest micro-ros-patformio, which uses arduino frame work. You may follow another wiki to build the robot base/ motor controller. https://github.com/hippo5329/linorobot2_hardware/wiki

hippo5329, 2024-08-18:
pc ~~wifi~~ robot computer eg rpi5 with lidar, camera --usb serial-- esp32/pico, mpu6050, tb6612..

flux0-0, 2024-08-24:
i have connect esp32 via wifi but if I dont plug the serial cable( esp32- laptop) , micro ros agent (udp 8888) not active. If i plug, it works again . Can you help me fix it ? My set up code about wifi " IPAddress agent_ip(192, 168, 100, 163);
    size_t agent_port = 8888;

     char ssid[] = "P402";
    char psk[]= "hocmakhongchoi";

    set_microros_wifi_transports(ssid, psk, agent_ip, agent_port);
    delay(10000);"  
my platform.ini "[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_port = /dev/ttyUSB0

board_microros_distro = humble
board_microros_transport = wifi
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
". I don't think there is a problem with the power supply, because when I change the power supply to another, Micro ros agent still doesn't work