micro-ROS/micro_ros_arduino #1820: Any update plans for esp32S3 or esp32C3?

winster-bai, 2024-08-21:
Hi, Is there any update plan for esp32S3 or esp32C3? These two chips are relatively excellent in the development of robots and smart devices, but the current micro ros does not seem to support them.

hippo5329, 2024-08-21:
If you switch to micro-ros-platformio arduino framework, the esp32s3 and esp32c3 support can added easily.

Follow my wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

The commit to add esp32s3 and esp32c3 support. https://github.com/hippo5329/micro_ros_arduino_examples_platformio/commit/d9215cad4de1e802923529c0083848d27886af4f

hippo5329, 2024-08-21:
However, the essential hardware for robots, pulse counter and PWM channels  are reduced or removed on these variants.

Pulse counters are used for wheel encoder. PWM channels are used for motor control.

esp32: 8 pulse counters, 16 PWM
esp32s3: 4 pulse counters, 8 PWM
esp32c3: no pulse counter, 6 PWM

So esp32c3 is not suitable for mobile robot control. esp32s3 is limited by it peripherals. esp32 is still the best overall.

nik1806, 2024-10-16:
A very simple solution is suggested here - https://www.youtube.com/watch?v=0R8VUPEkYhg.

Basically, renaming directory `esp32` to `esp32s3` in directory - `~/Arduino/libraries/micro_ros_arduino/src/`.

hippo5329, 2024-10-16:
Actually, the old esp32 gets more pwm and pcnt than the new variants. It may be critical to robotics. 

Esp32 : 16 pwm, 8 pcnt
Esp32s3 : 8 pwm, 4 pcnt
Esp32c3: 6 pwm, 0 pcnt

It is much easier and more power to use platformio and vscode.

hippo5329, 2024-10-16:
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/blob/main/platformio/platformio.ini

Support esp32s3, esp32c3.