micro-ROS/micro_ros_arduino #1829: Arduino Opta with Micro ROS Support

Controlone-dude, 2024-09-02:
I want to use Arduino Opta in one of my application, but I'm not getting any clues that it will compatible with Micro ROS or not.
If anyone worked with Arduino Opta, please give some clearity on it.

hippo5329, 2024-09-02:
Opta has only 8 inputs and 4 output relays. It is not useful for robotics, You should get started with esp32 or pico.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

gbr1, 2024-11-14:
hi @Controlone-dude,
theorically it should be compatible because it works on a STM32H7 such as Portenta H7 and Arduino Giga R1

gbr1, 2024-11-15:
@Controlone-dude please check this PR https://github.com/micro-ROS/micro_ros_arduino/pull/1867