micro-ROS/micro_ros_arduino #1852: micro ros parameters not working

AbdullahAhmedTaman, 2024-10-15:
## Issue template

- Hardware description:  ESP32 dev module
- RTOS: 
- Installation type:  Arduino IDE
- Version or commit hash:  Humble

#### Steps to reproduce the issue

#### Expected behavior
the micro ros parameter works
#### Actual behavior
the parameters doesn't appear in the parameter list in the terminal and the node is not working 
#### Additional information
I was trying to make 16 parameters for a thrusters node (8 parameters for ratios and 8 for direction) but when i run the node  the node it doesn't do any thing, it doesn't give me an error and the parameters doesn't show in the parameters list.

here is the full code:
```
#include <micro_ros_arduino.h>
#include <ESP32Servo.h>
#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc_parameter/rclc_parameter.h>
#include <rclc/executor.h>
#include <rcl/error_handling.h>
#include <geometry_msgs/msg/twist.h>
#include <std_msgs/msg/int64.h>  // Include message type for speed
#include <std_msgs/msg/int32_multi_array.h>  // Include message type for thrusters feedback
//#include <string.h>
#include <stdio.h>
#include <unistd.h>

rcl_subscription_t twist_subscriber;
rcl_subscription_t speed_subscriber;  // Subscriber for speed messages

rclc_parameter_server_t param_server;
// Setup parameter server options 
const rclc_parameter_options_t param_options = {
    .notify_changed_over_dds = true,
    .max_params = 16,
    .allow_undeclared_parameters = false,
    .low_mem_mode = false}; // to set all 16 once

rcl_node_t node;
rcl_allocator_t allocator;
rclc_support_t support;
rclc_executor_t executor;

// Timer for publishing thrusters feedback and for cheking parameters values
rcl_timer_t feedback_timer;

// Publisher for thrusters feedback
rcl_publisher_t thrusters_feedback_publisher;

std_msgs__msg__Int64 speed_msg;
geometry_msgs__msg__Twist twist_msg;
std_msgs__msg__Int32MultiArray thrusters_feedback_msg;

// Thrusters Pins
#define H_FRONT_LEFT_PIN    19
#define H_FRONT_RIGHT_PIN   18
#define H_BACK_LEFT_PIN     5     
#define H_BACK_RIGHT_PIN    17
     
#define V_FRONT_LEFT_PIN    16
#define V_FRONT_RIGHT_PIN   4
#define V_BACK_LEFT_PIN     2   
#define V_BACK_RIGHT_PIN    23

// Thrusters Numbers
[... 367 lines cut ...]
        &node,
        ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int64),
        "speed"
    ));

    // Initialize the timer for thrusters feedback
    const unsigned int feedback_timer_timeout = 100; // milliseconds
    RCCHECK(rclc_timer_init_default(&feedback_timer, &support, RCL_MS_TO_NS(feedback_timer_timeout), feedback_timer_callback));
    RCCHECK(rclc_timer_init_default(&feedback_timer, &support, RCL_MS_TO_NS(10000), parameter_timer_callback));

    // Create an executor to handle the callbacks
    RCCHECK(rclc_executor_init(&executor, &support.context, RCLC_EXECUTOR_PARAMETER_SERVER_HANDLES + 20, &allocator));  
    RCCHECK(rclc_executor_add_subscription(&executor, &twist_subscriber, &twist_msg, &thruster_twist_callback, ON_NEW_DATA));
    RCCHECK(rclc_executor_add_subscription(&executor, &speed_subscriber, &speed_msg, &speed_callback, ON_NEW_DATA));
    RCCHECK(rclc_executor_add_timer(&executor, &feedback_timer));
    RCCHECK(rclc_executor_add_timer(&executor, &feedback_timer));
    RCCHECK(rclc_executor_add_parameter_server(&executor, &param_server, NULL));
}

void loop() {
    // Continuously spin to handle messages and control thrusters
    RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(100)));
    delay(100); // Small delay to avoid overloading the CPU
}
```

hippo5329, 2024-10-16:
1. You will need to allocate memory for the multiarray message.

https://micro.ros.org/docs/tutorials/advanced/handling_type_memory/

2. You will need to increase the allocation of publishers, subscribers,  and services with user colcon meta. Parameters server needs 5 services. The default is too low.

https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/

You will need to rebuild library for arduino. This is much easier with platformio.

3. You may find the parameter server example in the micro-ros-demos project.

hippo5329, 2024-10-16:
https://github.com/micro-ROS/micro-ROS-demos/blob/jazzy/rclc/parameter_server/main.c

It is difficult to debug a large program. Especially when you are new. There are a lot of things to learn. You should build and test in small steps. Verify every basic function. Add new features one by one.

hippo5329, 2024-10-16:
You should get started with a small simple robot before you build your dream robot.

https://github.com/hippo5329/linorobot2_hardware/wiki#linorobot2_hardware-with-esp32-and-pico-support