micro-ROS/micro_ros_arduino #1854: Question about how to apply micro-ROS on FreeRTOS on teensy4.1 development board

ChoiYouJung, 2024-11-01:
## Issue template

- Hardware description:  Teensy4.1
- RTOS:  FreeRTOS
- Installation type:  Arduino IDE or MCUXpressoIDE
- Version or commit hash:  humble or Jazzy

#### Steps to reproduce the issue

I am trying to apply micro-ROS on FreeRTOS on the teensy4.1 development board.
**So, can I use the static library micro_ros_arduino for the micro-ROS library like I used in the bare metal environment on the teensy4.1 development board?**

#### Expected behavior
- Apply micro-ROS on FreeRTOS on teensy4.1 development board
- I would like to know how to apply micro-ROS on FreeRTOS on the teensy4.1 development board.

#### Actual behavior

#### Additional information
- The operating system of the Arduino IDE is Windows 10.

- The operating system of the MCUXpresso IDE is Windows 10.

hippo5329, 2024-11-01:
Just some background questions.

1. Are you familiar with ROS2 and micro-ROS? How do you run ROS2 on Windows? Why don't you run Ubuntu? Are you familiar with docker?
2. Why do you want to use teensy4.1? Why do you want to use micro-ROS on freertos on teensy4.1? Are you familiar with freertos on teensy? Are you familiar with micro-ros on freertos? Teensyduino core does not run freertos. The freertos port of teensy uses Teensystudio as drivers. It is not native freertos.

If you are a beginner to ROS2 and micro-ROS, you should use ubuntu instead of Windows. Although it is possible to run Jazzy on Windows, it is difficult for beginners. You can use docker to run ROS2. But you still need the knowledge of ubuntu linux.

If you can install a ubuntu 24.04 pc or VM, you should be able to learn ROS2 and micro-ros better than on  windows.

If you only have windows, you should learn vscode+ros2+docker and platformio.

If you are new to micro-ros, you should start with esp32 which is cheaper and easier than teensy4.1. The esp32 core runs freertos.

You may follow the wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

ChoiYouJung, 2024-11-04:
Thank you for your interest in my question.

1. ROS2 runs on Linux. And micro-ros-agent also runs on Linux.
We are only developing the Teensy 4.1 board (micro-ros-client) on Windows 10.

2. 
In the past, I have experience using Teensy 4.1 as bare-metal using the micro-ros-arduino library and example code from Github.
I am in a situation where I need to use FreeRTOS on the Teensy 4.1 board, and I wanted to know the possibility of using the library used in the micro-ros-arduino example after porting FreeRTOS, so I asked this question here.

hippo5329, 2024-11-04:
There is another rtos, chibios , https://github.com/greiman/ChRt
I am not sure if it works.

hippo5329, 2024-11-04:
[env:teensy41]
platform = teensy
board = teensy41
lib_deps =
    ${env.lib_deps}
    https://github.com/juliandesvignes/FreeRTOS-Teensy4.git

It can build blink_task and publisher with micro-ros and freertos. But I didn't test on hardware.