micro-ROS/micro_ros_arduino #1880: How to get an error message when rcl_ret_t is 1?

JamesTann, 2024-11-27:
## How to get an error message when rcl_ret_t is 1?

- Hardware: Arduino Portenta
- Version: humble

_Note: This is not an issue, just a question, so my apologies if this is not the right place to ask it._

I modeled this macro after the `RCCHECK` one used throughout the examples:

```
#define RCCHECK(fn)              \
  {                              \
    rcl_ret_t temp_rc = fn;      \
    if ((temp_rc != RCL_RET_OK)) \
    {                            \
      ros_error(temp_rc);        \
    }                            \
  }
```
Which calls this function:
```
void ros_error(const int temp_rc)
{
  Serial.println("Error code encountered from ROS: " + String(temp_rc));
  Serial.println(rcl_get_error_string().str);
  rcutils_reset_error();
  NVIC_SystemReset();
}
```

However, when I encounter an error, I see this garbled output:

Output:
[image]

What is the proper way of getting a more specific error message?

_Note: I am communicating to the Micro ROS Agent using ethernet, so my understanding is that I can safely using serial for debugging_

metanav, 2024-11-28:
Hi @pablogs9,

I am having similar issue so instead of creating a new issue I think it should write here. How can we enable the debugging messages? 

PS: I am already using the docker to generate precompiled library.

JamesTann, 2024-12-02:
Hello @pablogs9 ,
Is there a way to temporarily enable it for debugging purposes and then disable it again before deploying it? 

I am currently trying to debug an issue where `rclc_executor_spin_some` returns 1 after sending a service request, and finding it difficult to figure out where this error might be arising from (and I was hoping to fix it myself before asking here).

@metanav I am also building the library with Docker.

hippo5329, 2024-12-02:
@JamesTann Have you update the colcon meta to allocate memory for services? The default is very limited.

https://micro.ros.org/docs/tutorials/advanced/microxrcedds_rmw_configuration/

JamesTann, 2024-12-02:
Hi @hippo5329, 

Yes I have, although I was not sure what the requirements are for a service client. So I just tried increasing `MAX_SERVICES` and `MAX_CLIENTS` from 1 to 10 to try and get something working. I also have a subscriber and a parameter server on my node hence the other values in my `colcon.meta`:

```
        "rmw_microxrcedds": {
            "cmake-args": [
                "-DRMW_UXRCE_MAX_NODES=1",
                "-DRMW_UXRCE_MAX_PUBLISHERS=10",
                "-DRMW_UXRCE_MAX_SUBSCRIPTIONS=5",
                "-DRMW_UXRCE_MAX_SERVICES=10",
                "-DRMW_UXRCE_MAX_CLIENTS=10",
                "-DRMW_UXRCE_MAX_HISTORY=4",
                "-DRMW_UXRCE_TRANSPORT=custom"
            ]
        }
```
Should this be enough resources?