micro-ROS/micro_ros_arduino #1883: Unable to synchronize time between micro ros and ros2

flux0-0, 2024-11-29:
This is my code " https://github.com/flux0-0/test_microros_time/blob/main/main.cpp " . If I upload it to Esp32, I don't see anything printed in the serial. This is my ini file " [env:esp32doit-devkit-v1]
platform = espressif32@6.9.0
board = esp32doit-devkit-v1
framework = arduino

monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
monitor_port = /dev/ttyUSB0

board_microros_distro = humble
board_microros_transport = wifi
lib_deps =
    https://github.com/micro-ROS/micro_ros_platformio
    paulstoffregen/Time@^1.6 "

flux0-0, 2024-11-30:
Tks, I have tried a lot but I failed. Can you give me an example code about synchronous time in micro ros and ros2 with framework adruino ?

hippo5329, 2024-11-30:
Please check this,

https://github.com/hippo5329/linorobot2_hardware/blob/master/firmware/src/firmware.cpp#L246

flux0-0, 2024-12-02:
I've read and followed, but it seems my counter always starts at 0. This is my code "https://github.com/flux0-0/test_microros_time/blob/main/main.cpp" I hope you will read and help me point out any errors!

hippo5329, 2024-12-02:
There are a lot of things to learn. Coding, debugging, micro-ros re-connection, timer..etc. It takes time to learn. No short-cut. I would suggest follow the wiki and try out.

https://github.com/hippo5329/linorobot2_hardware/wiki

hippo5329, 2024-12-02:
You can try out with a bare esp32. No real robot or sensor is needed.

Then start over with the re-connection example. Change the message to a dummy stamped one. The add time sync. Do these step by step.

flux0-0, 2024-12-11:
Thanks for your help, I fixed it successfully

closed 2024-12-11 by flux0-0