micro-ROS/micro_ros_arduino #1885: Jazzy release - Arduino IDE or PlatformioIO - availability in the official libraries lists

rgruberski, 2024-11-29:
Hi! Would you like to add a precompiled library for Jazzy to the release list? As @EnzoGhisoni asked in the closed issue (#1851). I use a remote platform for tests, so I can't use the default branch or something custom-made. I need to have access to the official release. I need to have a precompiled release on the website: https://www.arduinolibraries.info/libraries/micro_ros_arduino

Without a release, using it with platforms like Arduino IDE or PlatformioIO from the official libraries list is impossible. The only way is to use it as a custom add-on library, but in my case, I can't do that.

Many thanks for considering my request :)

arne48, 2025-01-24:
Friendly ping, if a release for the Arduino library is possible soon.

arne48, 2025-10-01:
Hello @4ntn,
That's great, thank you very much :)