micro-ROS/micro_ros_arduino #1895: Micro_Ros_Arduino Not Compiling on Arduino Due

Robo52, 2025-01-29:
- Hardware description: 
- 
- Arduino Due (Atmel SAM3X8E) running standard Arduino environment (no RTOS)
- Raspberry Pi 4 running Ubuntu 22.04 for the micro-ROS Agent.
- 
- RTOS: 
- 
- Installation type: 
- micro_ros_setup on the RPI4B
- micro_ros_arduino@2.0.7-humble on the Arduino Due (installed via .zip in Arduino Library Manager)
- 
- Version or commit hash: 
- Humble

#### Steps to reproduce the issue

Installed micro_ros_arduino@2.0.7-humble in the Arduino IDE.

Try to compile this Arduino CPP code in the Arduino IDE: 

`#include <Arduino.h>
#include <micro_ros_arduino.h>

#include <rcl/rcl.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>
#include <std_msgs/msg/int32.h>

#include "MicroRosIntegration.h"

// External variables
extern int pwm[3];
extern volatile int position[3];
extern volatile unsigned long angle[3];

// External movement functions
extern void AllForward(int, int, int);
extern void AllReverse(int, int, int);
extern void TurnLeft();
extern void TurnRight();
extern void Calibrate();
extern void StopAll();

// micro-ROS global handles
rcl_subscription_t subscriber;
std_msgs__msg__Int32 cmd_msg;

rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;

// Callback for incoming commands
void command_callback(const void *msgin)
{
  const std_msgs__msg__Int32 *msg = (const std_msgs__msg__Int32 *)msgin;
  int command = msg->data;

  switch(command) {
    case 1:
      AllForward(pwm[0], pwm[1], pwm[2]);
      break;
    case 2:
      AllReverse(pwm[0], pwm[1], pwm[2]);
      break;
    case 3:
      TurnLeft();
      break;
    case 4:
      TurnRight();
      break;
    case 5:
      Calibrate();
      break;
    case 6:
      StopAll();
      break;
    default:
      // No-op
      break;
  }
}

void InitMicroRos()
{
  // Setup micro-ROS transport on Arduino (SerialUSB)
  set_microros_transports();

  allocator = rcl_get_default_allocator();
  rclc_support_init(&support, 0, NULL, &allocator);
  rclc_node_init_default(&node, "spherewalker_node", "", &support);

  // Create subscriber
  rclc_subscription_init_default(
    &subscriber,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "/spherewalker_cmd"
  );

  // Create executor
  rclc_executor_init(&executor, &support.context, 1, &allocator);
  rclc_executor_add_subscription(&executor, &subscriber, &cmd_msg, &command_callback, ON_NEW_DATA);

}

void SpinMicroRos()
{
  // Let micro-ROS process incoming messages
  rclc_executor_spin_some(&executor, RCL_MS_TO_NS(10));
}
`

With this heaader: 

`#ifndef MICRO_ROS_INTEGRATION_H
#define MICRO_ROS_INTEGRATION_H

void InitMicroRos();
void SpinMicroRos();

#endif`

#### Expected behavior

I expect to be able to compile the code and then connect the Arduino DUE to the RPI4B

#### Actual behavior

I get these errors:

/home/rockinlab/.cache/arduino/sketches/693122C21CBEE9609C004E70FEAA903B/sketch/MicroRosIntegration.cpp.o: In function `set_microros_transports':
/home/rockinlab/Arduino/libraries/micro_ros_arduino/src/micro_ros_arduino.h:40: undefined reference to `rmw_uros_set_custom_transport'
/home/rockinlab/.cache/arduino/sketches/693122C21CBEE9609C004E70FEAA903B/sketch/MicroRosIntegration.cpp.o: In function `InitMicroRos()':
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:71: undefined reference to `rcutils_get_default_allocator'
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:72: undefined reference to `rclc_support_init'
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:73: undefined reference to `rclc_node_init_default'
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:81: undefined reference to `rosidl_typesupport_c__get_message_type_support_handle__std_msgs__msg__Int32'
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:81: undefined reference to `rclc_subscription_init_default'
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:84: undefined reference to `rclc_executor_init'
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:85: undefined reference to `rclc_executor_add_subscription'
/home/rockinlab/.cache/arduino/sketches/693122C21CBEE9609C004E70FEAA903B/sketch/MicroRosIntegration.cpp.o: In function `SpinMicroRos()':
/home/rockinlab/Arduino/SphereWalker1/MicroRosIntegration.cpp:92: undefined reference to `rclc_executor_spin_some'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

#### Additional information

hippo5329, 2025-01-30:
Please remove the micro-ros zip release and use git version of humble or jazzy branch. Due should be supported.

Or you may use platformio.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

pio run -e due

Robo52, 2025-01-30:
@hippo5329 So I use this? This is what I have been using. Or am I supposed to git clone somewhere? Sorry really new to micro-ros getting a bit confused.

[image]

Robo52, 2025-01-30:
@hippo5329 I also am not sure if this makes a difference but I am not using the Pi with micro-ros-agent to upload to the Due. I am on another PC that does not have micro-ros-agent.

hippo5329, 2025-01-30:
I would suggest follow the wiki and try out the basic example. You may do it on your pc or pi.

hippo5329, 2025-01-30:
https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

pio run -e due -t upload

Robo52, 2025-01-30:
I am trying the example but when I try to build in the IDE on VScode I get this error: 

.pio/libdeps/due/micro_ros_arduino/src/cortex-m3/libmicroros.a(librmw-validate_full_topic_name.c.obj): In function `rmw_validate_full_topic_name':
validate_full_topic_name.c:(.text.rmw_validate_full_topic_name+0xf8): undefined reference to `__ctype_ptr__'
.pio/libdeps/due/micro_ros_arduino/src/cortex-m3/libmicroros.a(librmw-validate_node_name.c.obj): In function `rmw_validate_node_name':
validate_node_name.c:(.text.rmw_validate_node_name+0x88): undefined reference to `__ctype_ptr__'
.pio/libdeps/due/micro_ros_arduino/src/cortex-m3/libmicroros.a(librcl-validate_topic_name.c.obj): In function `rcl_validate_topic_name':
validate_topic_name.c:(.text.rcl_validate_topic_name+0x1cc): undefined reference to `__ctype_ptr__'

closed 2025-01-30 by Robo52