micro-ROS/micro_ros_arduino #1904: Where is source code?

asukiaaa, 2025-02-07:
I tried to add function to resolve the following issue but I cannot find source of `rclc_executor_add_subscription`.
https://github.com/micro-ROS/micro_ros_arduino/issues/1508

Is this library closed code in .a file?
https://github.com/micro-ROS/micro_ros_arduino/blob/jazzy/src/esp32/libmicroros.a

If it is open, I want to see the code.

hippo5329, 2025-02-08:
https://github.com/ros2/rclc/blob/rolling/rclc/src/rclc/executor.c#L207

BTW, I would suggest switch to platformio. Follow the wiki, https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

asukiaaa, 2025-02-08:
Thank you for the information.
I understood that this library is wrapper for rclc on arduino.

> BTW, I would suggest switch to platformio. 

Why?
I usually use platformio but I'm not sure why you suggest so.

hippo5329, 2025-02-08:
Arduino uses pre-built micro-ros library, while platformio builds micro-ros library. It will be easier to customize micro-ros on platformio. Arduino is good only if you have to run windows.Platformio cannot build micro-ros on windows.

The wiki is a guide to run arduino micro-ros examples on platformio. It should be easier than arduino.

asukiaaa, 2025-02-08:
I understood.
Thank you for the explanation.