micro-ROS/micro_ros_arduino #1915: Arduino Due is Limited to 110 Hz

midasama3124, 2025-02-22:
- Hardware description: Arduino Due
- RTOS: None
- Installation type: precompiled libraries on Arduino IDE
- Version or commit hash: ROS2 Humble

#### Steps to reproduce the issue
I am just running the simple publisher provided in the examples. The only difference is that I removed all delays and dialed the publisher timer. I did that to see what the maximum sampling frequency is.

`#include <micro_ros_arduino.h>

#include <stdio.h>
#include <rcl/rcl.h>
#include <rcl/error_handling.h>
#include <rclc/rclc.h>
#include <rclc/executor.h>

#include <std_msgs/msg/int32.h>

rcl_publisher_t publisher;
std_msgs__msg__Int32 msg;
rclc_executor_t executor;
rclc_support_t support;
rcl_allocator_t allocator;
rcl_node_t node;
rcl_timer_t timer;

#define LED_PIN 13

#define RCCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){error_loop();}}
#define RCSOFTCHECK(fn) { rcl_ret_t temp_rc = fn; if((temp_rc != RCL_RET_OK)){}}

void error_loop(){
  while(1){
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
  }
}

void timer_callback(rcl_timer_t * timer, int64_t last_call_time)
{  
  RCLC_UNUSED(last_call_time);
  if (timer != NULL) {
    RCSOFTCHECK(rcl_publish(&publisher, &msg, NULL));
    msg.data++;
  }
}

void setup() {
  set_microros_transports();
  
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);  
  
  delay(2000);

  allocator = rcl_get_default_allocator();

  //create init_options
  RCCHECK(rclc_support_init(&support, 0, NULL, &allocator));

  // create node
  RCCHECK(rclc_node_init_default(&node, "micro_ros_arduino_node", "", &support));

  // create publisher
  RCCHECK(rclc_publisher_init_default(
    &publisher,
    &node,
    ROSIDL_GET_MSG_TYPE_SUPPORT(std_msgs, msg, Int32),
    "micro_ros_arduino_node_publisher"));

  // create timer,
  const unsigned int timer_timeout = 1;
  RCCHECK(rclc_timer_init_default(
    &timer,
    &support,
    RCL_MS_TO_NS(timer_timeout),
    timer_callback));

  // create executor
  RCCHECK(rclc_executor_init(&executor, &support.context, 1, &allocator));
  RCCHECK(rclc_executor_add_timer(&executor, &timer));

  msg.data = 0;
}

void loop() {
  RCSOFTCHECK(rclc_executor_spin_some(&executor, RCL_MS_TO_NS(1)));
}`

#### Expected behavior
I would like to reach a sampling frequency at least around 400-500 Hz.

#### Actual behavior
It is not going past 110 Hz even with this simple publisher.

#### Additional information
I have tried updating the baud rate in Arduino/libraries/micro_ros_arduino-2.0.5-[DISTRO]/src/default_transport.cpp (as suggested [here](https://github.com/micro-ROS/micro_ros_arduino/issues/1816)).
I also tried recompiling the libraries by upgrading the compiler (as done [here](https://github.com/micro-ROS/micro_ros_arduino/pull/490)) but I couldn't manage to make it work on Arduino IDE. I kept getting multiple errors.

hippo5329, 2025-02-23:
What is your host computer? Are your host running native ubuntu 22.04? Due is slow 84MHz and memory is quite limited. You should not use Due for micro-ROS. You may try pico/pico2 or esp32. They are much faster and have more memory. They have more useful peripherals for robotics than most arduino boards family.

You may switch to platformio/vscode. It is more powerful and flexible. Follow this wiki and try out the re-connection example.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki

pio run -e due -t upload

hippo5329, 2025-02-23:
Another question, why do you want this high message rate?

midasama3124, 2025-02-23:
Hi, thanks for the quick reply.

I have used two desktop computers as hosts: One using Ubuntu 24.04 natively (Jazzy), and another running Ubuntu 22.04 on a Windows VM (Humble).

I forgot to mention but I already tried using platformio as well. And even though the simple publisher gets built and uploaded to the Arduino Due, when I run the micro_agent, the publisher doesn't get acknowledged (it keeps waiting for a connection).

I noticed that unplugging and reconnecting the microboard would solve this issue when using the Arduino IDE, but it doesn't for the case of platformio unfortunately. But now that I think about it, I have only tried this alternative on the VM, could that affect the output?

Regarding the sampling rate, I am developing a lower-limb exoskeleton and the reason I'm using this board is for collecting encoder data to capture velocity measurements. The velocity values provided by my motors ended up being too noisy for the controller I use, so now I need to use an additional sensor. I am using Arduino Due because it is the only board supported by micro_ros that is available at my workplace. But I am still open to getting another board as you mentioned or just somehow integrating these encoders to the CAN network that my motors use already.

Sorry for the long background, but the high sampling rate is because my device would be assisting within a reduced time window while the person is wearing it (i.e., less than a second). This means the closer all my states are to real-time the better.

Thanks for the help again,
Miguel

hippo5329, 2025-02-23:
I didn't mean the simple publisher. I meant the re-connection example. I would suggest try out the wiki.

For new design, you should use Jazzy.

VM on Windows is not recommended for micro-ros. You should use native ubuntu or docker on other linux distro.

Desktop computers are fast. They can reach 400-500Hz topic rate on my tests with pico/esp32.

What kind of motor/encoder do you use? What do you mean by additional encoder? I would suspect design/wiring issues. Micro-ROS is often used for motor control. Control loop in 20-50Hz should be fine for most robotic usages.

Pico/pico2 and esp32 support hardware pulse counter for encoder. They are cheap and popular.

hippo5329, 2025-02-23:
https://articulo.mercadolibre.com.co/MCO-1315241239-raspberry-pi-pico-rp2040-original-regletas-sin-soldar-_JM

midasama3124, 2025-02-23:
I also tried the reconnection example, and it was not acknowledged by the agent either. But again, I only tried it on the VM, I will try with the host running Jazzy natively later and will let you know.

Yeah, I don't think the host computer is the problem. Every other node running either actuation or sensor measurements runs at high frequencies with no issue. The only bottleneck right now is this encoder being captured by the Arduino Due.

The motors we use are the Cubemars AK 10-9, and I don't think it is a wiring problem since the position measurement is clean and accurate. However, the velocity measurement reported by these motors is noisy and affects the performance of my controller. That's why I'm trying another external encoder (not the one built into the motor) to get that velocity.

A control loop running at 20-50Hz would be too low for my case, unfortunately. We only have a period of around 1 second to actuate, so having only 50 samples to compute the desired torque on the motors would not be ideal.

I will try running platformio on the host running Ubuntu natively. If it does not work, I will purchase the boards you recommend or capture the encoder data with some CAN-based board so the entire device uses the same communication protocol.

Thanks.

hippo5329, 2025-02-24:
Here is the ros2 driver,

https://github.com/OpenFieldAutomation-OFA/cubemars_hardware

hippo5329, 2025-02-24:
For your application, VM on Windows is NOT recommended. You should use native ubuntu.

midasama3124, 2025-02-24:
The motors are already working. I used a PeakCan adapter instead. Thanks for taking the time to look it up though.

I am using another encoder for the velocity reading because computing it from the position reported by the motor may induce instabilities in my controller.

I just tried using platformio on the computer running Ubuntu (ROS2 Jazzy) natively and I see the same as in the VM.

I am attaching the console outputs I get for both the reconnection and the simple publisher examples. As you can see, the micro_agent is not acknowledging the board and there are no new topics. 

The final screenshot shows the way it works when building and uploading the code using Arduino IDE.

Thanks,
Miguel

[image]
[image]
[image]

closed 2025-02-26 by midasama3124