micro-ROS/micro_ros_arduino #1980: 'eprosima::fastcdr::exception::NotEnoughMemoryException'   what():  Not enough memory in the buffer stream

YongJunYou, 2025-05-19:
## Issue template

- Hardware description: Teensy 4.0
- Installation type:  micro_ros_setup
- Version or commit hash: humble

#### Steps to reproduce the issue
1. Flash micro-ROS firmware to microcontroller (Teensy, Arduino, etc.).
2. Connect device via USB (/dev/ttyACM0).
3. Run the micro-ROS Agent:
(https://youtu.be/SV7DeB7kDZs?si=zt7mhgJJjIvvHUZl)
   ```bash
   ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0

#### Expected behavior
micro-ROS Agent runs continuously without crashing, enabling communication with the microcontroller.

#### Actual behavior

ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0
[1747650676.305213] info     | TermiosAgentLinux.cpp | init                     | running...             | fd: 3
[1747650676.305449] info     | Root.cpp           | set_verbose_level        | logger setup           | verbose_level: 4
[1747650677.266568] info     | Root.cpp           | create_client            | create                 | client_key: 0x4A0AE1BE, session_id: 0x81
[1747650677.266734] info     | SessionManager.hpp | establish_session        | session established    | client_key: 0x4A0AE1BE, address: 0
[1747650677.279361] info     | ProxyClient.cpp    | create_participant       | participant created    | client_key: 0x4A0AE1BE, participant_id: 0x000(1)
[1747650677.279939] info     | ProxyClient.cpp    | create_topic             | topic created          | client_key: 0x4A0AE1BE, topic_id: 0x000(2), participant_id: 0x000(1)
[1747650677.280375] info     | ProxyClient.cpp    | create_publisher         | publisher created      | client_key: 0x4A0AE1BE, publisher_id: 0x000(3), participant_id: 0x000(1)
[1747650677.614655] info     | ProxyClient.cpp    | create_datawriter        | datawriter created     | client_key: 0x4A0AE1BE, datawriter_id: 0x000(5), publisher_id: 0x000(3)
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
[ros2run]: Aborted

#### Additional information
Ubuntu 22.04
source build

I used the default micro_ros_publisher.ino code. Could this be an issue specific to the Teensy 4.0?

hippo5329, 2025-05-20:
You may switch to platformio and follow the wiki.

https://github.com/hippo5329/micro_ros_arduino_examples_platformio/wiki#use-teensy41

    pio run -e teensy40 -t upload
    ros2 run micro_ros_agent micro_ros_agent serial --dev /dev/ttyACM0

YongJunYou, 2025-05-26:
I followed the wiki, but in the end, it didn’t work—just like the Arduino IDE.

hippo5329, 2025-05-27:
Humble will be EOL in one year.

hippo5329, 2025-05-27:
I installed a fresh ubuntu 22.04 and ros2 humble on bare metal. There is no issue with teensy40.

closed 2025-09-30 by 4ntn